klausahrenberg/WThermostatBeca #88: targetTempertature no longer reflects schedule set temperature

julezman, 2020-03-26:
My thermostats currently report the incorrect targetTemperature (via MQTT, smartThings etc) but show the correct targetTemperate on the display and operate the schedule and heating correctly.  They seem to show the last set manual temperature.  The actual temperature reports and updates correctly via MQTT etc.  Updates to the schedule and any other setting changes via MQTT work as expected.  All 5 thermostats have all worked OK for 2-3 months on v1.00 and previously reported targetTemperature in line with schedules.  All 5 suddenly started doing this overnight 2 days ago with no changes to the settings as far as I'm aware.  schedulesMode is still set to auto as it always has been.  I've tried rebooting each Thermostat and even reflashed firmware on a few of the devices and use @fashberg v1.03b firmware on another (with updated topics for that thermostat) and still no luck.  While investigating, I have noticed there is an issue with worldtimeapi currently but clock still reports correct time for all thermostats but can't see how this would affect this functionality. Any ideas what I've (inadvertadently) done or what has happened and how I can fix?

P.S - Thanks for the great firmware and recent work on forks too!

julezman, 2020-03-26:
OK - suddenly started working again.  And worldtimeapi.org is back up and running.  It would seem that either the lack of timezone, validTime set to false as a result of worldtimeapi being down causes this issue.  Is there any way to manually override these settings to avoid this behaviour?  Will take a look at code amnd see if anything jumps out.

austwhite, 2020-03-26:
I never had any issues with mine using worldtimeapi.org.
I have constantly been running it 24/7 in testing, abusing with multiple temp changes etc and it has always been spot on.
I use Folkes 1.02b.
Be aware 1.01b, 1.02b and 1.03b are not from Klause.  They are updated porta from Folkes.

julezman, 2020-03-26:
> I never had any issues with mine using worldtimeapi.org.
> I have constantly been running it 24/7 in testing, abusing with multiple temp changes etc and it has always been spot on.
> I use Folkes 1.02b.
> If you have a version that works for you, stick to that. There is no need to update thermostat firmware if it works properly 🙂

Thanks for your reply.  Interesting that you had no issues with worldtimeapi over the last day or so as at least one other person has reported it down (https://serverfault.com/questions/1008352/is-there-a-free-json-time-api-besides-worldtimeapi-org).  For me even when manually navigating to the worltimeapi.org URL on different devices I got a page stating the website was down for maintenance.

As I said all was working fine with 1.00 with worldtimeapi up until yesterday.  Also, as I said I only tried reflashing and other firmware in case this helped to resolve and am aware that 1.03b comes from Folkes hence me tagging them!

fashberg, 2020-03-26:
FYI: worldtimeapi was just down for nearly one day. Seems to be back right now.
One of my thermostats was set to UTC because of this. Happy to live not to far east/west to Greenich :) 

[image]

austwhite, 2020-03-26:
@fashberg Would it be possible to modify the code to allow manual entry of timezones rather than relying on a web page?  Timezones as per unix/linux format are pretty well documented and easy to find.

austwhite, 2020-03-26:
> Thanks for your reply. Interesting that you had no issues with worldtimeapi over the last day or so as at least one other person has reported it down

Maybe mine didn't check the NTP while the site was down :).   It's back now which is the main thing.
Maybe a manual timezone or manual way to set time would be a good option for this firmware.

julezman, 2020-03-26:
> @fashberg Would it be possible to modify the code to allow manual entry of timezones rather than relying on a web page? Timezones as per unix/linux format are pretty well documented and easy to find.
> Manual time zone would also eliminate errors which may be caused by VPN connection or by service providers that do not accurately reflect the location in the IP address allocation.

Agree - this would be great.  Its great that it can do this automatically, but should there be a fall back\default option people can set too?  Or code can cache\store\retain previous timezone that was retrieved and use this if API is down?  Finally, perhaps calling the API less frequently will help reduce load on API server?

fashberg, 2020-03-27:
> Agree - this would be great. Its great that it can do this automatically, but should there be a fall back\default option people can set too? Or code can cache\store\retain previous timezone that was [...]

Hi @austwhite and @julezman,

yeah, i will try to do this. Currently the Daylight Saving calculation completely relies on the worldtimeapi response, so it's not just the timezone. Must be implemented something like this: https://tasmota.github.io/docs/#/Commands?id=timestd  (TimeSTD/TimeDST)

I've found one bug related to targetTemperature in WBecaDevice.h
`changed = ((changed) || (WProperty::isEqual(targetTemperatureManualMode, newValue, 0.01)));`
The isEqual has to be negated. Currently the new state is only reported if the temperature has _not_ been changed.

Will publish my 1.04b soon, together with great logging to MQTT improvement, configurable as a new pseudo device (things/logging) and also configurable verbosity at webgui.

Kind Regards from corona homeoffice

austwhite, 2020-03-27:
@fashberg 
Look forward to it.  So far I have been enjoying using your firmware.  The improvements on Klaus's firmware have made it much more usable

ramon-roca, 2020-03-29:
@fashberg 

I've been now testing 1.03b this weekend. I saw a few things while trying to read/send via MQTT.

I've seen targetTemperature with schedulesMode on giving the result as it was when was previosly set manually, Not sure how to reproduce. While sending targetTemperature swirches also to shedulesMode off (manual).
Also on BAC Thermostat, I can't set systemMode or mode (cool|heat|fan_only), neither fanMode.

fashberg, 2020-03-31:
Hey guys!

I've just relased my 1.04b - https://github.com/fashberg/WThermostatBeca - with support of fixed timezone and Daylight Saving Rules.
See: https://github.com/fashberg/WThermostatBeca/blob/master/Configuration.md#4-configure-clock-settings.

Don't forget to check all settings after Upgrade.

@ramon-roca : can you try 1.04 and set logging to trace (on webpage) and then show a little bit of the logs while trying to set fanMode or systemMode. You can open a new issue  on https://github.com/fashberg/WThermostatBeca 

Kind Regards

Folke

ramon-roca, 2020-03-31:
Hi folke, 

Can't open issues at https://github.com/fashberg/WThermostatBeca there is no such option there.
Just uploaded 1.05b and enabled trace
Tested systemMode and fanMode; seems that works like a charm. 

**GOOD JOB!**

I'll keep testing and upload to the other devices and give feedback..

I can attach trace if you want it...

THANKS

fashberg, 2020-04-01:
Hi @ramon-roca 
> I'll keep testing and upload to the other devices and give feedback..
> I can attach trace if you want it...

Good to hear - if everything works no trace neccessary 

I have enabled issues page at the fork-page.

Kind Regards
Folke

closed 2020-04-03 by klausahrenberg