klausahrenberg/WThermostatBeca #89: Thermostat control not working, only clock function present (continued from #78)

poedator, 2020-04-03:
Hello, @klausahrenberg,

I tried 1.02 firmware but the problem persists.
device model: BHT-002GBLW
the symptoms are same as described in my post here: https://github.com/klausahrenberg/WThermostatBeca/issues/78#issuecomment-602276571

Basically the device connects and syncs clock (screen shows correct time) but it fails to interact with any thermostat functions.  
`http://[IP ADDRESS]/things/thermostat/properties/state` shows `{"state":"off"}` in all cases, even when the thermostat is on manually...

bessarabov, 2020-04-03:
I have BHT-002GBLW (I'm now on 1.00, haven't tested the new firmware yet). Everything works fine. But I'm using mqtt and don't use http.

I've checked what my thermostat says via http. But the url you have specified return 404 in my case:

```
pi@raspberrypi:~ $ curl http://192.168.31.40/things/thermostat/properties/state
pi@raspberrypi:~ $ curl http://192.168.31.40
[{"name":"thermostat","href":"/things/thermostat","@context":"https://iot.mozilla.org/schemas","@type":["Thermostat"],"properties":{"temperature":{"title":"Actual","type":"number","readOnly":true,"unit":"celsius","@type":"TemperatureProperty","href":"/things/thermostat/properties/temperature"},"targetTemperature":{"title":"Target","type":"number","unit":"celsius","multipleOf":0.50,"@type":"TargetTemperatureProperty","href":"/things/thermostat/properties/targetTemperature"},"deviceOn":{"title":"Power","type":"boolean","@type":"OnOffProperty","href":"/things/thermostat/properties/deviceOn"},"schedulesMode":{"title":"Schedules","type":"string","enum":["off","auto"],"@type":"ThermostatModeProperty","href":"/things/thermostat/properties/schedulesMode"}}}]

pi@raspberrypi:~ $ curl -v curl http://192.168.31.40/things/thermostat/properties/state
* Expire in 0 ms for 6 (transfer 0x1d9b7c0)
* Expire in 1 ms for 1 (transfer 0x1d9b7c0)
[LOTS OF LINES  "Expire in" SKIPPED]
* Expire in 1 ms for 1 (transfer 0x1d9b7c0)
* Could not resolve host: curl
* Expire in 2 ms for 1 (transfer 0x1d9b7c0)
* Closing connection 0
curl: (6) Could not resolve host: curl
* Expire in 0 ms for 6 (transfer 0x1d9b7c0)
*   Trying 192.168.31.40...
* TCP_NODELAY set
* Expire in 200 ms for 4 (transfer 0x1d9b7c0)
* Connected to 192.168.31.40 (192.168.31.40) port 80 (#1)
> GET /things/thermostat/properties/state HTTP/1.1
> Host: 192.168.31.40
> User-Agent: curl/7.64.0
> Accept: */*
>
< HTTP/1.1 404 Not Found
< Content-Type: text/html
< Content-Length: 0
< Connection: close
<
* Closing connection 1
pi@raspberrypi:~ $
```

klausahrenberg, 2020-04-03:
Hi poedator, sorry to hear. I have to say that I didn't recognize your issue. Moreover, I have the same model. 

> situation: only updates of time and config posted to mqtt at start and no regular temperature updates. Besides, the thermostat is not even reacting on incoming mqtt messages

Please give me your complete json result of: `http://[IP ADDRESS]/things/thermostat/properties`. Also listen at your MQTT browser to all topics #, if a message with mcucommand - unknown arrives. Please check.

Your device must communicate in another way. The devices has another internal controller (MCU) that is not touched by my firmware. This MCU communicates with the ESP controller. My ESP firmware asks at start the MCU and expect 5 parameters at least. Before these 5 params are not received, the ESP firmware will not send any MQTT status to prevent that incomplete/wrong information will sent. I think here is the failure because your clock messages arrives which are independent from thermostat. So we have to find out which paramaters are not available at your device or have a different message format than at my device. The 5 required params are: deviceOn, targetTemperature, schedulesMode, ecoMode and locked. Does your device has all these functions. 3 of the 5 you can check by the json above.

poedator, 2020-04-03:
Hi, Klausahrenberg,

It could be the case that Tuya slightly changed MCU protocol in my device version...

Here is output from `http://[IP ADDRESS]/things/thermostat/properties`:

`{"idx":"thermostatbeca_1","ip":"192.168.1.219","firmware":"1.02","temperature":0.00,"targetTemperature":0.00,"deviceOn":false,"schedulesMode":"","state":"off"}`

Here is output from `http://[IP ADDRESS]/things/thermostat` (JSON formatted:
I noticed that it has no `ecoMode` and `locked` properties.

`"name":"thermostat",
   "href":"/things/thermostat",
   "@context":"https://iot.mozilla.org/schemas",
   "@type":[
      "Thermostat"
   ],
   "properties":{
      "temperature":{
         "title":"Actual",
         "type":"number",
         "readOnly":true,
         "unit":"celsius",
         "@type":"TemperatureProperty",
         "href":"/things/thermostat/properties/temperature"
      },
      "targetTemperature":{
         "title":"Target",
         "type":"number",
         "unit":"celsius",
         "multipleOf":0.50,
         "@type":"TargetTemperatureProperty",
         "href":"/things/thermostat/properties/targetTemperature"
      },
      "deviceOn":{
         "title":"Power",
         "type":"boolean",
         "@type":"OnOffProperty",
         "href":"/things/thermostat/properties/deviceOn"
      },
      "schedulesMode":{
         "title":"Schedules",
         "type":"string",
         "enum":[
            "off",
            "auto"
         ],
         "@type":"ThermostatModeProperty",
         "href":"/things/thermostat/properties/schedulesMode"
      },
      "state":{
         "title":"State",
         "type":"string",
         "readOnly":true,
         "enum":[
            "off",
            "heating",
            "cooling"
         ],
         "@type":"HeatingCoolingProperty",
         "href":"/things/thermostat/properties/state"
      }
   }
}`

poedator, 2020-04-03:
I opened the device. The board looks exactly like yours (as on picture here: https://github.com/klausahrenberg/WThermostatBeca/blob/master/Flashing.md). 
On the TYWE3S (ESP) module it says  `model:2.22.03.02754`. Yours says `model:2.22.03.01771`. Just in case it matters...

klausahrenberg, 2020-04-04:
So the json says that your MCU did not send the status. Yes maybe they changed the protocol, but then you should get some unknown messages over MQTT at least. Was this device working with the original firmware? If yes, when und where did you bought this device? If protocol really changed, I will order one for investigation.

austwhite, 2020-04-04:
@bessarabov 
In post two I noticed an error with the curl command.  You tried to curl to a host named curl.
I don't think you should have the second curl after the -v

Energy88, 2020-04-06:
> So the json says that your MCU did not send the status. Yes maybe they changed the protocol, but then you should get some unknown messages over MQTT at least. Was this device working with the [...]

Tthe same problem, propreties output only:
`{"idx":"thermostatbeca_12590593","ip":"192.168.1.25","firmware":"1.03","temperature":24.00,"targetTemperature":20.00,"deviceOn":true,"schedulesMode":"off","state":"heating"}`

temperatures are right, deviceOn is also correct, but in my case state is always "heating"

i have bought it [here](https://aliexpress.ru/item/4000203375726.html?invitationCode=Q3BrdzBhZ2NMZm1LZm9UNTd6VHVCT29BOUJaN3p1MHBvK3NUSEs3TUJhWi9PaEx4NExOUmhBPT0&mb=Xv1WVNPQdPpW4a6&srcSns=Copy+to+Clipboard&tid=white_backgroup_101&tt=sns_Copy&image=U5f2504c8fd7a4a438ff4a888e8539e0ct.jpg&aff_request_id=0ccf2acfab784a5dac45e67e830d16b8-1586196160536-01579-_ePtrEN&spreadType=socialShare&aff_platform=default&sk=_ePtrEN&aff_trace_key=0ccf2acfab784a5dac45e67e830d16b8-1586196160536-01579-_ePtrEN&businessType=ProductDetail&templateId=white_backgroup_101&title=%D0%94%D0%B5%D0%BB%D0%B0%D0%B9%D1%82%D0%B5+%D0%BF%D0%BE%D0%BA%D1%83%D0%BF%D0%BA%D0%B8+%D0%BD%D0%B0+AliExpress+%D1%81+%D0%BA%D1%83%D0%BF%D0%BE%D0%BD%D0%B0%D0%BC%D0%B8+%D0%B4%D0%BB%D1%8F+%D0%BD%D0%BE%D0%B2%D0%B8%D1%87%D0%BA%D0%BE%D0%B2&platform=AE&terminal_id=9d3b14da4a2748bfa3eb85c8135d7d44)

bessarabov, 2020-04-06:
> temperatures are right, deviceOn is also correct, but in my case state is always "heating"

You need to make hardware modification get info about relay state — https://github.com/klausahrenberg/WThermostatBeca/issues/17#issuecomment-552078026

Energy88, 2020-04-09:
> > temperatures are right, deviceOn is also correct, but in my case state is always "heating"
> 
> You need to make hardware modification get info about relay state — [#17 (comment)](https://github.com/klausahrenberg/WThermostatBeca/issues/17#issuecomment-552078026)

Thanks, it works! Is some way to get floortemperature data? thermostat shows only internal themeperature. Also, i can toggle "schedulesMode" by MQTT, but hardware doesn't react on it.

bessarabov, 2020-04-09:
> Thanks, it works! Is some way to get floortemperature data? thermostat shows only internal themeperature. Also, i can toggle "schedulesMode" by MQTT, but hardware doesn't react on it.

I'm not interested in using thermostat "schedulesMode", so I don't know anything about it.

I had no problem with the floor temp. I was getting it from the very beginning:
```

Message 1 received on thermostatbeca_13610097/things/thermostat/properties at 10:05 PM:
{
    "idx": "thermostatbeca_13610097",
    "ip": "192.168.31.40",
    "firmware": "1.00",
    "temperature": 7,
    "targetTemperature": 3,
    "deviceOn": true,
    "schedulesMode": "off",
    "ecoMode": false,
    "locked": true,
    "floorTemperature": 13
}
```

klausahrenberg, 2020-04-09:
Hi, if floor heating is not present in json, you selected the wrong model in settings. If you select the "floor heating" model bac-002-gblw in settings than you will have the parameter, otherwise not.

poedator, 2020-04-19:
> Hi, if floor heating is not present in json, you selected the wrong model in settings. If you select the "floor heating" model bac-002-gblw in settings than you will have the parameter, otherwise not.

I checked - it always had `BHT-002-GBLW` selected as 'thermostat model. The other alternative is `BAC-002-ALW` which is not for floor heating, as I understand.

It reports temperature, just it is zero, not the target temp...
from http://192.168.1.219/things/thermostat/properties: `{"idx":"thermostatbeca_1","ip":"192.168.1.219","firmware":"1.02","temperature":0.00,"targetTemperature":0.00,"deviceOn":false,"schedulesMode":"","state":"heating"}`

austwhite, 2020-04-19:
@poedator 
You are correct.  The ALW model is for the variable fan coil / heat cool model.
You need to choose the GBxx model for floor heating

poedator, 2020-04-19:
@austwhite you are correct, the floor temperature is not displayed in  `/things/thermostat/properties`. On display I only see one temperature, which may be room temperature, not floor. It is same even with sensor disconnected. Sensor is working, 10 kOhm. 
Could be some electric malfunction - but it did work with original beca firmware.
BTW, I always had GBxx model selected on config.

austwhite, 2020-04-19:
@klausahrenberg 
I've noticed this same behaviour with some testing.
Is the floor sensor (the optional.one) displayed in the returned data?  If so, can you advise how to activate it?

austwhite, 2020-04-25:
Okay,  Floor temp is showing in my list now.  I reflashed the firmware and now it shows.  I don't use the sensor so it just shows 0.
I chose Floor Heating (BHT-002-GBLW) as the Thermostat model on the Thermostat page.
It shows both in JSON format and as a separate topic if using separate topic mode.

@poedator  Maybe can try reflashing?

klausahrenberg, 2020-09-23:
Please test latest version and tell me, if problem persists.

poedator, 2020-10-08:
Hi, @klausahrenberg!
thanks for persistence in firmware development. I tried flashing from 1.03 to 1.18 via web interface (http://[IP]/firmware?) but it fails with error `Not Enough Space \n ESP reboots now...` This is strange because FW is not that large. 
Please suggest how to deal with this. I hesitate to solder connections and use programmer but if you tell me that there is no other way - I'll do it.

poedator, 2020-10-08:
Update: I was unable to flash 1.17, 1.10 but was successful with 1.04, which did not fix my issue. Shall I flash all versions in sequence?

poedator, 2020-10-08:
Update: flashed 1.07 over 1.05 and then 1.10. over 1.07 but then it stopped connecting to wifi...

klausahrenberg, 2020-10-08:
Hi, you are too fast. This issue was a bug in this old version because I set the wrong compiler options. Please flash this fork firmware: https://github.com/fashberg/WThermostatBeca/releases/download/v1.15-fas/wthermostat-minimal-1.15-fas.bin After that you can flash to the current firmware version. After that you will not have this issue anymore, I corrected the wrong compiler settings at new version.

PS: You have to reconfigure all settings between flashing the different versions

poedator, 2020-10-09:
Hi, @klausahrenberg, I finally managed to install 1.18 - the problem is gone  - it reports current and target temp just fine!  Thanks for fixing  - BTW what was it? 

I noticed that the thermostat manages environment temperature, but not directly the floor temperature. Is there a way to make it manage floor temp directly?

closed 2020-10-09 by poedator

klausahrenberg, 2020-10-12:
Good to know that it's working now.

> I noticed that the thermostat manages environment temperature, but not directly the floor temperature. Is there a way to make it manage floor temp directly?

The usage of floor sensor you can select directly at the thermostat, see the manual: https://support.gw-ec.com/upload/qa/2018/11/19/3019d10504894c7327d18fc07389bbe6.pdf

poedator, 2020-10-12:
thanks again! I set option 4 to 'Ou' and it is controlling just floor temperature now.
I understand, that this function is managed by the other processors of the thermostat. It is really weird that they had to use 3 (!) quite powerful processors to do the job would be easy even to a single Arduino.