klausahrenberg/WThermostatBeca #91: Schedules get/set over MQTT doesn't work

Hedgehog57, 2020-04-04:
FW: 1.02
Thermostat: GCWL

I can get schedules over http:
[image]
I cannot get schedules over MQTT:

[image]

I cannot set schedules over MQTT. Cannot remember how I have did it last time with FW 1.0, but i think I have set schedules over MQTT.

Hedgehog57, 2020-04-04:
Oh, I found a working configuration.
There is an error in the documentation.
To set schedules:
```
<your_mqtt_topic/thermostat/<your_set_topic>/schedules
```
But I still don't understant how to get schedules over MQTT. ATM I can check current schedules only over HTTP.

klausahrenberg, 2020-04-05:
Hi,
Schedules getting and setting is still possible, but my dcumentation was not up-to-date, thanks for hint. Now it's corrected:
https://github.com/klausahrenberg/WThermostatBeca#2-schedules
Request actual schedules via message with empty payload to 
`<your_mqtt_topic>/thermostat/<your_state_topic>/schedules`
e.g.: `<your_mqtt_topic>/thermostat/properties/schedules`
Schedules can be modified via MQTT. Send a payload structure with all schedules or only the parts you want to modify to
`<your_mqtt_topic>/thermostat/<your_set_topic>/schedules`
e.g.: `<your_mqtt_topic>/thermostat/set/schedules`
PS: There was still a minor bug: After setting of schedules, the device didn't respond the whole schedules for confirmation. It's fixed with version 1.03

austwhite, 2020-04-05:
@klausahrenberg 
Thank you so much for your work on this and the super quick responses.
I just want you to know it is very much appreciated.

Hedgehog57, 2020-04-05:
@klausahrenberg 
Thanks a lot!  Now (1.03) getting schedules works too.
Issue can be closed.
P.S. Upgrading from 1.02 to 1.03 didn't reset all settings and conection to wifi wasn't lost. Very well.

klausahrenberg, 2020-04-05:
Thanks for checking

closed 2020-04-05 by klausahrenberg