klausahrenberg/WThermostatBeca #92: Bht 3000 series

truxntrax, 2020-04-09:
Hi,

Do you think your firmware could work on the but 3000 series thermostat? I tasmota'd it and tried your firmware but tasmota responded "firmware not compatible" ( or similar message). 
The chips looks very similar to your version.
[image]
Many thanks

John

austwhite, 2020-04-09:
It would be at your own risk to try it on that thermostat as the TUYA module may send different codes.

Make sure you keep a backup of the original firmware in case you need to revert back.

truxntrax, 2020-04-09:
Hi,
I have a backup of the original firmware, tuyaconvert method makes a backup. I'll solder the wires onto the chip and give it a try. I will let you know my results..

Thanks

John

klausahrenberg, 2020-04-22:
Any results?

truxntrax, 2020-04-22:
Pains me to say it. But bad soldering and I killed the board.... So, yet undecided what to do next. Might buy a stat like yours. ( Need to practice my soldering skills and get that right!/)

John

austwhite, 2020-04-22:
Those boards are easy to kill with bad soldering.
I always update the firmware with Tuya Convert method.  It seems to work with most.
I do like to do the hardware mod to get the status though, but at least that is only to joints to solder.....
I feel your pain.  I killed my first one with a slightly to hot soldering iron lifting pads

klausahrenberg, 2020-04-27:
> But bad soldering and I killed the board...
Hmm, this never happened to me :-) I'm joking, sorry to hear - I know that's frustrating. I close this issue. If you buy the same model, let me know.

closed 2020-04-27 by klausahrenberg

pflegefall, 2020-06-08:
fyi: Upload of WThermostat_1.11.bin via Tuya convert worked seamlessly on Bht 3000.
Acutally I am integrating the device via MQTT to OPENHAB.

For an additional installation an implementation of Heating/Cooling functionality with a potential override function via MQTT would be great. Is it possible?

BR,
pflegefall

tomchovanec, 2020-07-01:
I can confirm that v 1.11 worki on BHT-3000. 
I just had troubles with uploading intermediate firware (second try worked).
Just FYI, BHT-3000 can be put into "smartconfig" mode by holding UP arrows key for 8s when device is off.