klausahrenberg/WThermostatBeca #93: Installing firmware on 3439845871846 thermostat

tantalusblank, 2020-04-13:
Product page for thermostat [here](https://totaltiles.co.uk/wifi-electric-underfloor-heating-thermostat-white.html)

Hi,
This may be a completely wrong use of 'issues' as I am trying to use this outside of the scope of the project, in that it's not even a Beca thermostat. It does however have a TYWE3S chip on board, so I thought I'd give it a go. Unfortunately it hasn't worked, and I'm getting "unknownMCU" messages on the MQTT broker, so the messages from the MCU must be fundamentally different.

Not sure if it's worth mentioning this model as definitely not compatible?

klausahrenberg, 2020-04-15:
To get 'unknownMCU' messages isn't a bad sign. Did you receive any parameters, like 'temperature' or did you not receive any valid value? If you get an 'unknownMCU' every time you press a function on the device, a support of device is possible. The device model seems to be ET-81W, I don't own this device. But if you can give me the MCU commands + the action you did on the device, maybe we get this working.

klausahrenberg, 2020-04-15:
If you don't receive a state record via MQTT, check state json result at: `http://[IP ADDRESS]/things/thermostat/properties`

tantalusblank, 2020-04-18:
Amazing - thanks so much for giving this a look. There does seem to be a bit of a pattern, but it doesn't seem to respond straight away to changes on the thermostat.

**I've monitored the topic using MQTT explorer, and get the following:**

[image]

**This is a little later on the same topic:**

[image]

**This is the 'error' topic on the broker:**

[image]

**And this is the result I get from "http://[IP ADDRESS]/things/thermostat/properties"**

[image]

Hopefully this sheds some light on whether this is possible or not

klausahrenberg, 2020-04-19:
Ok, at least your device sends a temperature. Value 'cd' is 205 > 20.5 C and 'e6' is 230 > 23 C. This is different from my model where the values must only divided by 2 instead of 10. With this my firmware can't deal at the moment.
However, the unknown command is only 1. That's not enough to assume, how much costs it will be to make it fit for your thermostat. Please try the following firmware which will send all received mcu commands via mqtt and give me all received commands. But please send it in text form, not as screen capture.
[WThermostat_1.03_logmcu.zip](https://github.com/klausahrenberg/WThermostatBeca/files/4498185/WThermostat_1.03_logmcu.zip)

tantalusblank, 2020-04-19:
I flashed the "logmcu" firmware and configured it (and checked the box to send properties over MQTT - not sure if this was necessary).

**These are the messages (oldest first) on the **"thermostatbeca_4766904/thermostat/failure"** topic after turning it on:**

{"unknownMCU":"55 aa 01 07 00 08 06 02 00 04 00 00 01 09"}
{"unknownMCU":"55 aa 01 07 00 08 07 02 00 04 00 00 00 3b"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 c8"}
{"unknownMCU":"55 aa 01 07 00 08 09 02 00 04 00 00 00 c8"}
{"unknownMCU":"55 aa 01 07 00 05 0a 01 00 01 00"}
{"unknownMCU":"55 aa 01 07 00 05 0b 01 00 01 00"}
{"unknownMCU":"55 aa 01 07 00 08 0c 02 00 04 00 00 00 00"}
{"unknownMCU":"55 aa 01 07 00 08 0d 02 00 04 00 00 00 00"}
{"unknownMCU":"55 aa 01 07 00 05 0e 04 00 01 02"}
{"unknownMCU":"55 aa 01 07 00 08 0f 02 00 04 00 00 00 00"}
{"unknownMCU":"55 aa 01 07 00 08 10 02 00 04 00 00 00 00"}
{"unknownMCU":"55 aa 01 07 00 08 11 02 00 04 00 00 00 00"}
{"unknownMCU":"55 aa 01 07 00 08 12 02 00 04 00 00 00 00"}
{"unknownMCU":"55 aa 01 07 00 05 13 04 00 01 01"}
{"unknownMCU":"55 aa 01 07 00 05 14 04 00 01 01"}
{"unknownMCU":"55 aa 01 07 00 05 15 01 00 01 00"}
{"unknownMCU":"55 aa 01 07 00 08 16 02 00 04 00 00 00 00"}
{"unknownMCU":"55 aa 01 07 00 08 17 02 00 04 00 00 00 02"}
{"unknownMCU":"55 aa 01 07 00 05 19 04 00 01 00"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 cd"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 c8"}
{"unknownMCU":"55 aa 01 07 00 08 05 02 00 04 00 00 01 04"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 cd"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 c8"}
{"unknownMCU":"55 aa 01 07 00 08 05 02 00 04 00 00 00 ff"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 cd"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 c8"}
{"unknownMCU":"55 aa 01 07 00 08 05 02 00 04 00 00 00 fa"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 cd"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 c8"}

**Changing the set temperature:**

{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 91"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 8c"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 87"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 82"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 7d"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 82"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 87"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 8c"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 91"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 96"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 9b"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 a0"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 a5"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 aa"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 cd"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 af"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 b4"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 b9"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 be"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 c3"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 c8"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 cd"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 d2"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 d7"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 dc"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 e1"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 e6"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 eb"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 f0"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 f5"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 fa"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 ff"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 01 04"}
[... 12 lines cut ...]
{"unknownMCU":"55 aa 01 07 00 08 05 02 00 04 00 00 00 f0"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 cd"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 c8"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 cd"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 c8"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 cd"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 c8"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 cd"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 c8"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 cd"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 c8"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 cd"}

**Turning power off and on:**

{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 c8"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 32"}
{"deviceOn_x01":"55 aa 01 07 00 05 01 01 00 01 00"}
{"unknownMCU":"55 aa 01 07 00 05 19 04 00 01 00"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 96"}
{"deviceOn_x01":"55 aa 01 07 00 05 01 01 00 01 01"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 cd"}
{"unknownMCU":"55 aa 01 07 00 08 05 02 00 04 00 00 00 f5"}
{"unknownMCU":"55 aa 01 07 00 08 05 02 00 04 00 00 00 f0"}
{"unknownMCU":"55 aa 01 07 00 08 05 02 00 04 00 00 00 f5"}

klausahrenberg, 2020-04-20:
Please test version 1.04 and report the unknown messages. In device configuration select your model ET-81W. 
Experimental support for other devices ET-81W (still for testing only). These devices has other temperature values and other schedules. Please report any unknown messages and which function on thermostat was selected. So far, following parameters are expected to work: deviceOn, temperature, targetTemperature, floorTemperature. Schedules will not work. Unclear functions, please report: Time synchronization, schedulesMode, ecoMode, locked

To be clear: We have to make several rounds with this to get your model working - I don't own the device, so I need your input. 
I see at your logs that some commands are still unknown for me. The "command byte" is always at the 7. position of the MCU command. For example command byte of "55 aa 01 07 00 08 05 02 00 04 00 00 00 f5" is 05. In v 1.04 I assumed the follwing commands:
05 is temperature, 02 is target temperature, 08 is floor temperature - Please check
Still unknown at your device
Command 17 and 19: Which function did you select?

tantalusblank, 2020-04-20:
Absolutely - happy to do whatever testing is needed. 

### Some notes on 1.04

- **deviceON** seems to work correctly except for on initial startup. The thermostat always seems to boot in the ON state, but the initial MQTT messages show it as OFF. If the heating is turned OFF, then ON again, it will update to the correct value. Not sure if this is to do with MQTT retained messages.

- **temperature** and **floorTemperature** are both giving readings that make sense, but are the wrong way around. This was checked in two ways: firstly each sensor was selected on the thermostat (rather than both), and the readouts checked against MQTT. Secondly, when heating, the "temperature" value increased much faster than the "floorTemperature" value.

- **targetTemperature** is reading correctly, though when the device is OFF it reads at "5" on MQTT (the set temp is the same whether ON/OFF on the thermostat)

As expected, nothing else is coming through. The state seems to be "heating" no matter what the temperature is. Will this need a hardware modification to work?

Example message:
{"idx":"thermostatbeca_4766904","ip":"192.168.1.102","firmware":"1.04","temperature":24.50,"targetTemperature":15.00,"deviceOn":true,"schedulesMode":"","ecoMode":false,"locked":false,"floorTemperature":23.00,"state":"heating"}

### Command bytes

These values were recorded using the **WThermostat_1.03_logmcu.bin** firmware.

When the topic is cleared and the thermostat is booted up, it seems to rapidly go through commands: 06 ,07 ,08, 09, 0a, 0b, 0c, 0d, 0e, 0f, 10, 11, 12, 13, 14, 15, 16, 17, 19. These messages were all sent at the same time after booting up:

{"unknownMCU":"55 aa 01 07 00 08 06 02 00 04 00 00 01 09"}
{"unknownMCU":"55 aa 01 07 00 08 07 02 00 04 00 00 00 3b"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 d7"}
{"unknownMCU":"55 aa 01 07 00 08 09 02 00 04 00 00 00 d7"}
{"unknownMCU":"55 aa 01 07 00 05 0a 01 00 01 00"}
{"unknownMCU":"55 aa 01 07 00 05 0b 01 00 01 00"}
{"unknownMCU":"55 aa 01 07 00 08 0c 02 00 04 00 00 00 00"}
{"unknownMCU":"55 aa 01 07 00 08 0d 02 00 04 00 00 00 00"}
{"unknownMCU":"55 aa 01 07 00 05 0e 04 00 01 02"}
{"unknownMCU":"55 aa 01 07 00 08 0f 02 00 04 00 00 00 00"}
{"unknownMCU":"55 aa 01 07 00 08 10 02 00 04 00 00 00 00"}
{"unknownMCU":"55 aa 01 07 00 08 11 02 00 04 00 00 00 00"}
{"unknownMCU":"55 aa 01 07 00 08 12 02 00 04 00 00 00 00"}
{"unknownMCU":"55 aa 01 07 00 05 13 04 00 01 01"}
{"unknownMCU":"55 aa 01 07 00 05 14 04 00 01 01"}
{"unknownMCU":"55 aa 01 07 00 05 15 01 00 01 00"}
{"unknownMCU":"55 aa 01 07 00 08 16 02 00 04 00 00 00 00"}
{"unknownMCU":"55 aa 01 07 00 08 17 02 00 04 00 00 00 08"}
{"unknownMCU":"55 aa 01 07 00 05 19 04 00 01 00"}

**This is what happened when I set a holiday for one day:**

{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 32"}
{"unknownMCU":"55 aa 01 07 00 05 03 04 00 01 00"}
{"unknownMCU":"55 aa 01 07 00 08 0c 02 00 04 00 00 00 01"}

**And then back to no holiday:**

{"unknownMCU":"55 aa 01 07 00 08 0c 02 00 04 00 00 00 00"}
{"unknownMCU":"55 aa 01 07 00 05 03 04 00 01 01"}
{"targetTemperature_x02":"55 aa 01 07 00 08 02 02 00 04 00 00 00 96"}

**Setting on hold for one hour:**

{"unknownMCU":"55 aa 01 07 00 05 03 04 00 01 02"}
{"unknownMCU":"55 aa 01 07 00 08 0d 02 00 04 00 00 00 3c"}
{"unknownMCU":"55 aa 01 07 00 08 05 02 00 04 00 00 00 fa"}
{"unknownMCU":"55 aa 01 07 00 08 0d 02 00 04 00 00 00 3b"}

**Exactly one minute later:**

{"unknownMCU":"55 aa 01 07 00 08 0d 02 00 04 00 00 00 3a"}

**Exactly another minute later:**

{"unknownMCU":"55 aa 01 07 00 08 0d 02 00 04 00 00 00 39"}

**Removing hold:**

{"unknownMCU":"55 aa 01 07 00 08 0d 02 00 04 00 00 00 39"}
{"unknownMCU":"55 aa 01 07 00 08 0d 02 00 04 00 00 00 3a"}
{"unknownMCU":"55 aa 01 07 00 08 0d 02 00 04 00 00 00 3b"}
{"unknownMCU":"55 aa 01 07 00 08 0d 02 00 04 00 00 00 00"}
{"unknownMCU":"55 aa 01 07 00 05 03 04 00 01 01"}

**Changing from °C to F:**

{"unknownMCU":"55 aa 01 07 00 05 0a 01 00 01 01"}
{"unknownMCU":"55 aa 01 07 00 08 06 02 00 04 00 00 00 4c"}
{"unknownMCU":"55 aa 01 07 00 08 09 02 00 04 00 00 00 47"}

**Back to °C:**

{"unknownMCU":"55 aa 01 07 00 05 0a 01 00 01 00"}
{"unknownMCU":"55 aa 01 07 00 05 19 04 00 01 01"}
{"unknownMCU":"55 aa 01 07 00 08 05 02 00 04 00 00 00 f5"}
{"unknownMCU":"55 aa 01 07 00 08 08 02 00 04 00 00 00 d7"}
{"unknownMCU":"55 aa 01 07 00 05 19 04 00 01 00"}

**Changing control backlash:**

{"unknownMCU":"55 aa 01 07 00 05 13 04 00 01 02"}
{"unknownMCU":"55 aa 01 07 00 05 14 04 00 01 02"}

**Changing power rating (from 0 to 100W):**

{"unknownMCU":"55 aa 01 07 00 08 16 02 00 04 00 00 00 64"}

**Changing to room sensor:**

{"unknownMCU":"55 aa 01 07 00 05 0e 04 00 01 01"}
{"unknownMCU":"55 aa 01 07 00 05 0e 04 00 01 00"}

**Changing to probe sensor:**

{"unknownMCU":"55 aa 01 07 00 05 0e 04 00 01 01"}

**Changing back to both:**

{"unknownMCU":"55 aa 01 07 00 05 0e 04 00 01 02"}

**No messages:** 
- Navigating menus without selecting anything
- Clock changes (although it seems to trigger an 05 command when changed)
- Changing schedule
- Adjusting frost protection
- Floor probe temperature limit change
- Delay output change
- Backlight brightness change

There isn't an eco-mode available on the model.

Hopefully this helps

klausahrenberg, 2020-04-20:
Ok, I will check. One question: The clock of thermostat is not setted after first startup, right?

tantalusblank, 2020-04-20:
I believe there's a battery to retain settings on power loss but I'm not sure whether time /date is included in that. If you're asking whether I changed anything on the thermostat after booting - I did not - those messages were all without any interaction, just from boot up.

klausahrenberg, 2020-04-22:
Version 1.05 now corrected temperature<>floorTemperature and supports parameter schedulesMode for your model with values "auto|hold|holiday". Please test.

tantalusblank, 2020-04-25:
I've now had a chance to test out 1.05:

- I'm able to turn off and on from home assistant, well as changing set temperature. This is the main thing I needed to do and finally having this control is awesome - thanks!

- The floor and room temperature values seem to be correct.

- schedulesMode shows as blank ("") until a setting is changed. It reacts to hold and holiday correctly, though it doesn't react to a 'set' message through MQTT (e.g. topic: "kitchenheating/thermostat/set/schedulesMode" payload: "holiday"). I tried sending an empty payload to "kitchenheating/thermostat/properties/schedules" to request the schedules but I'm not sure what topic it would come back on - I can't see anything pop up on the topics. I might just not be understanding this part of it so will try and have another look early next week.

Bettman66, 2020-04-25:
If you don't use Mozilla iot, you can also test with Tasmota. Tasmota supports a scripting language with which I can now completely control my Beca thermostat.
[Picture](https://forum.iobroker.net/assets/uploads/files/1587456434935-tasmota_tuya_thermostat.png)
You only have to enter your values in the script.
`
>D
p:ha=1
p:lo=0
p:ec=0
p:soll=0
pin5=0
target=0
temp=0
tempf=0
>E
soll=TuyaReceived#DpType2Id2/2
temp=TuyaReceived#DpType2Id3/2
tempf=TuyaReceived#DpType2Id102/2
if chg[temp]>0 {
=>publish stat/%topic%/RESULT {"istwert":"%1temp%"}
}
if chg[tempf]>0 {
=>publish stat/%topic%/RESULT {"floor":"%1tempf%"}
}
if chg[target]>0 {
soll=target/2
}
>F
if chg[ha]>0 {
=>TuyaSend4 4,%0ha%
=>publish stat/%topic%/RESULT {"Script":">ha=%0ha%"}
}
if pin5!=pin[5] {
pin5=pin[5]
=>publish stat/%topic%/RESULT {"state":"%0pin5%"}
}
if chg[soll]>0 {
target=soll*2
=>TuyaSend2 2,%target%
=>publish stat/%topic%/RESULT {"Script":">soll=%1soll%"}
}
if chg[ec]>0 {
=>TuyaSend1 5,%0ec%
=>publish stat/%topic%/RESULT {"Script":">ec=%0ec%"}
}
if chg[lo]>0 {
=>TuyaSend1 6,%0lo%
=>publish stat/%topic%/RESULT {"Script":">lo=%0lo%"}
}
>W
ck(pin5 "RelaisState on/off   ")
Sollwert:{m} %1soll% °C
Istwert:{m} %1temp% °C
Floor:{m} %1tempf% °C
sl(30 60 target "15" "Sollwert" "30")
bu(ha "Hand" "Auto")
bu(ec "Eco" "Normal")
bu(lo "Lock" "Unlock")`

jsponz, 2020-04-26:
@tantalusblank, how is the status? What is missing? 

I do really like the color screen. Is it worthy?

tantalusblank, 2020-05-30:
@jsponz sorry for the slow reply - seems pretty worthy to me! Trying to integrate it with Home Assistant's MQTT climate configuration and not quite there - but it's responding to published messages.

@Bettman66 thanks for the suggestion - I've only tried Tasmota as an install on NodeMCU or Sonoffs, and trying to get my head around how this would work. Do you mean load Tasmota on to the thermostat itself?