klausahrenberg/WThermostatBeca #95: 0.5 degrees issue

pewi121, 2020-04-17:
Hi Klaus,

I flashed your really great firmware 1.03 on many BECA BHT-1000 via Tuya OTA. 
Works like a charm.  I am very happy. Thank you for that!
There is only one little minor issue: 
With MQTT I can not change the degrees in 0.5 steps but in 1.0 steps only, although the thermostat supports 0.5 steps. If I do a 0.5 step change, it jumps to next full degree up or down.
On the touchscreen it is no problem to change 0.5 degrees.

thanks for support
Peter

pewi121, 2020-04-17:
And one more question: Without the HW mod, "state" always returns "heating".
Correct?

thank you !

fashberg, 2020-04-18:
Hi Peter,

Do you use colomn or dot as decimal separator? E.g. 20.5 vs. 20,5.

You have to use dot!

Yes, without HW mod the wifi module does not know the state of the relays.
The relays are getting controlled by the second microcontroller (MCU) depending on target temperature, room/floor temperature, schedeler, hysteresis, etc.

Kind regards

Folke

pewi121, 2020-04-18:
Hi Folke,
i use a number item an this seems to be in correct format using dot (eg 22.5).
Example: When I change from 23.0 to 22.5 log file shows: 
------------
Item 'Thermostat_Target_Temperature' received command 22.5
Thermostat_Target_Temperature predicted to become 22.5
Thermostat_Target_Temperature changed from 23.00 to 22.5
Thermostat_Target_Temperature changed from 22.5 to 22.00
------------
So it dows two changes not only one as the expected.
This happens only downwards. If I increase temperature the .5 steps it works fine.
I think this is an openhab issue and not related to the firmware.
Will try to solve with openhab...

thank you

fashberg, 2020-04-19:
Hey,

Hmm, who controls that messages?
Can you try sending 22.5 directly with MQTT publish to thermostat while  openhab (i guess) shut down.

See this page for some examples but take care on using the right MQTT Topic, my version uses fixed topics cmnd/stat like tasmota

https://github.com/fashberg/WThermostatBeca/blob/master/README.md#modifying-parameters-via-mqtt

Kind Regards

Folke

pewi121, 2020-04-19:
Hi Folke,

I use MQTT with Mosquitto.
I tried your sugestion and can see that using MQTT.fx there is no issue sending 0.5 step degrees up or down. 
So it seems to be an issue in the setpoint command of OpenHAB.

thank you for help...

pewi121, 2020-04-19:
Solved:
I changed the OpenHAB locale from DE_UTF8 to EN_UTF8.
Now the degree setting works as expected with .5 steps up and down ...

closed 2020-04-19 by pewi121