klausahrenberg/WThermostatBeca #96: "schedulesMode" property sends wrong format?

pewi121, 2020-04-19:
Hi,

I have another issue:
Sending and receiving commands from BECA thermostat works fine, but in OpenHAB logfile I get:
--------
Command '"off"' not supported by type 'OnOffValue': No enum constant org.eclipse.smarthome.core.library.types.OnOffType."off"
--------
Command '"auto"' not supported by type 'OnOffValue': No enum constant org.eclipse.smarthome.core.library.types.OnOffType."auto"
--------
While other statements deliver there value like this:
wz/thermostat/properties/targetTemperature  18.00
wz/thermostat/properties/deviceOn  true
the schedulesMode command delivers values like: "auto" and "off".
I can verify this behavior in MQTT.fx.

The command is working, even it post a warning in the log every time...

thank you
Peter

klausahrenberg, 2020-04-21:
I think this is the expected behavior for parameter schedulesMode. According to the Readme, it can have the values "auto" or "off". Your message is caused by OpenHAB, which seems to expect values "true" or "false" for an OnOffValue. Please check, if you can reconfigure OpenHAB to handle the values given by the thermostat. Maybe there is an enumeration type, that would be the right one to choose.

pewi121, 2020-04-22:
Hi Klaus,
you are (of course) right. The readme tells me the possible values, but I wonder why all values are send without "" except the schedules value which uses "auto" and "off" instead of auto and off.

radokristof, 2020-06-01:
And how can I disable the schedules completely? I don't need them, I want to control it manually and now on 1.11 seems that it turns back to schedule automatically when the next period start time is reached (I did not changed it, the default values can be seen on the web interface).

klausahrenberg, 2020-06-01:
Hi, you can disable this optional feature in the web interface > Configure Thermostat. Deactivate the option 'Auto mode from manual mode at next schedule period change'

radokristof, 2020-06-05:
Thanks it worked!

closed 2020-06-05 by klausahrenberg