klausahrenberg/WThermostatBeca #97: How look the Display over the Keys?

wlan4you, 2020-04-20:
Hello,
wow, great Project. I want to buy this BHT-002-GBLW.
My old AEG electrical control is very boring and the TIME is running up after some weeks.

Great are, that i can put TASMOTA on the Firmware.
But, how is looking the front Display and the Keys?
Can you make some Picture from this ?

Thank You
Stefan
Or you can send it to me by email.
Greetings from Germany

fashberg, 2020-04-26:
hi! 
Check out here: https://github.com/fashberg/WThermostatBeca/blob/master/docs/device.jpg

changing the firmware will not change the behaviour on the display and keys. 
Not possible. Why: https://github.com/fashberg/WThermostatBeca/#hardware

Kind Regards from Munich

Folke

klausahrenberg, 2020-04-27:
Nothing to add to comments from @fashberg

closed 2020-04-27 by klausahrenberg