klausahrenberg/WThermostatBeca #99: Bht-002 MOES and home assistant

edellamaria, 2020-04-24:
Hi Klaus,
I really feel stupid with uch a question, but don't find the way to start with.
Flashed your last firmware, but now can't integrate the device into home assistant, I suppose using MQTT, right?
The other point is that you say that your fw is suitable fir bht-002GCLW but once in device conf there is only a GBLW model...can I use that?
I'm sorry for my basic questions, hope you understand that also we beginners need help to grow into this fantastic world :-)

here is what ip responds at the moment:

[{"name":"thermostat","href":"/things/thermostat","@context":"https://iot.mozilla.org/schemas","@type":["Thermostat"],"properties":{"temperature":{"title":"Actual","type":"number","readOnly":true,"unit":"degree celsius","@type":"TemperatureProperty","href":"/things/thermostat/properties/temperature"},"targetTemperature":{"title":"Target","type":"number","unit":"degree celsius","multipleOf":0.50,"@type":"TargetTemperatureProperty","href":"/things/thermostat/properties/targetTemperature"},"deviceOn":{"title":"Power","type":"boolean","@type":"OnOffProperty","href":"/things/thermostat/properties/deviceOn"},"schedulesMode":{"title":"Schedules","type":"string","enum":["auto","off"],"@type":"ThermostatModeProperty","href":"/things/thermostat/properties/schedulesMode"},"state":{"title":"State","type":"string","readOnly":true,"enum":["off","heating","cooling"],"@type":"HeatingCoolingProperty","href":"/things/thermostat/properties/state"}}}]

Thanks in advice

austwhite, 2020-04-24:
You have to manually add it as an MQTT platform thermostat.  The data can be sent in JSON as per default, or you can set this firmware to send each item individually.

I havent updated all my MQTT for the new topics yet.  I'll be working on that today.

the HA documentation is here: https://www.home-assistant.io/integrations/climate.mqtt/

edellamaria, 2020-04-24:
Wow austenite...I was not prepared for an immediate answers...THANK YOU!
Ok, with your tips it's my job and late night's to try now...but it could be (sarcastic) that I'll come back for further help.
Again thanks.

fashberg, 2020-04-24:
Try my fork with homeassistant auto-discovery: https://github.com/fashberg/WThermostatBeca

fashberg, 2020-04-24:
2nd Question is also documented:
https://github.com/fashberg/WThermostatBeca
[..]
The BHT Version is for heating only. The BAC-Version has modes Cooling, Heating and Ventilation. The HBT-002-GA/GB/GC versions only differs in relais-wiring.

austwhite, 2020-04-24:
@fashberg I tried to flash your fork, but kept getting the error "not enough space" when trying to update from Klaus's 1.05.
I can't even flash Tasmota-minimal as it is also giving the same error.

Any suggestions?

bwired-nl, 2020-04-24:
was also my question, why is there no HTTP interface? I use the fork to get this

iMike78, 2020-04-24:
@fashberg I'm really appreciate for you hard work on this fork. 
I have only two questions: 

1. Could you put back the restart/reboot button to the web interface? Sometimes it's needed. I know, I can do it on the http://<thermostat_ip>/reset, but would be much easier on the web interface.
2. Is there any chance to show the uptime on the info page?

Cheers

austwhite, 2020-04-25:
Anyone had anyluck getting Home Assistant to actually send payload "true" when setting the thermostat to heat mode?
I can only say HA sending "heat" which is being ignored by the floor thermostat model as it expects "true" or "false" for deviceOn.

austwhite, 2020-04-25:
> @fashberg I tried to flash your fork, but kept getting the error "not enough space" when trying to update from Klaus's 1.05.
> I can't even flash Tasmota-minimal as it is also giving the same error.
> 
> Any suggestions?

I worked around this.  I flashed back to 1.01b and then was able to successfully flash 1.09.

austwhite, 2020-04-25:
> I can only see HA sending "heat" which is being ignored by the floor thermostat model as it expects "true" or "false" for deviceOn.

I found out what the issue is over previous firmware.  For the BHT-002-GBLW model, the `mode` attribute is removed.  This attribute use to show Heat or Off for the thermostat and Home Assistant was able to use this to set Heat or Off via the `mode_topic_command` MQTT topic.
Without this in place, Home Assistant is not able to send mode=heat as the `deviceOn` topic expects only true or false.

@klausahrenberg Is there any chance of adding the `mode` attribute back for the BHT-002-GBLW model?

NightNigel, 2020-04-25:
@austwhite Sorry to bother, but where to find 1.01b version of firmware? Looks like it had been deleted...

austwhite, 2020-04-25:
@NightNigel 
The older versions are not available for general download.
I have a copy of it saved from an earlier upgrade.

austwhite, 2020-04-25:
As a note, 1.09b-fas works great with HA and auto discovers properly too.

NightNigel, 2020-04-25:
@austwhite Can you somehow share 1.01b.bin? Found out that version 1.02+ give same "not enough space" error. And tuya-convert doesn't work either.

austwhite, 2020-04-25:
[fas101b.zip](https://github.com/klausahrenberg/WThermostatBeca/files/4532831/fas101b.zip)
I have attached a zip file to this, hopefully it works.  It has @fashberg version 1.01b and also @klausahrenberg 1.00 in the file.
These are outdated now obviously.  The setup information on the repositories will not correspond with how these old versions are setup

austwhite, 2020-04-25:
@klausahrenberg   Can you check why we get the Not Enough Space error with your current version?
at 371KByte size, it should have space as the thermostat is 1MByte flash size.   Is it the use of SPIFF's?

NightNigel, 2020-04-25:
@austwhite Workaround with 1.01b works. Thank you.

austwhite, 2020-04-25:
@NightNigel I should have warned you that settings get lost when you switch to the @fashberg firmware.  They must be stored differently to the @klausahrenberg firmware.
At least they are easy to configure mostly.

NightNigel, 2020-04-25:
@austwhite Yes, configuration was reseted. It's not a big deal. Logging though,  was set to "warning messages" by default. Thanks again.

klausahrenberg, 2020-04-25:
Hi, please help me with the mode issue and HA. Why it have to be named to 'mode' and can not be 'schedulesMode' like now? In general it's not a big deal to rename, I choosed it because the ventilation models have another parameter called 'systemMode' - and I wanted a clear naming. But I can rename the parameter to 'mode' and this mode has the values 'off' or 'auto'. Is this how HA could use this parameter?

klausahrenberg, 2020-04-25:
For the 'nor enough space' issue. The current version is at the limit with it's size. I'm working on it to reduce the size of the binary. Next version you will have some reduction of size. However, flashing to @fashberg version 1.09 is not directly possible at moment, because his version is slightly bigger than mine. As I said, I try to reduce the size.

austwhite, 2020-04-25:
@klausahrenberg  
Thanks so much for the quick response.
Home Assistant, for setting modes, uses a topic called mode_command_topic.   This topic will send one of 4 data options.  `heat`, `cool`, `fan`, `off`.  There is also a mode_state_topic which receives the data options.
It does not send or receive `true` or `false`.  
This means it cannot use the `deviceOn` topic in the current firmware as `deviceOn` expects `true` or `false`.
I tried using the Home Assistant topic `power_command_topic` but this seems to only send the off command as `false` in the current implementation of Home Assistant and will not send a "true" value, meaning the thermostat cannot be turned on via Home Assistant.  It also means Home Assistant would have to use Optimistic Mode to determine the state, which means it doesn't actually really know the current state and if the device was turned off manually at the thermostat, Home Assistant would not know it had been turned off.

@fashberg has added a `mode` attribute to his firmware, which allows Home Assistant to use the `mode_command_topic` to send `heat`, `cool`,` fan` or `off ` to the mode topic and this will then correctly activate the thermostat.  This `mode` attribute also allows Home Assistant to always know the state of the thermostat as it is a read and write attribute and readable with the Home Assistant `mode_state_topic` function.

The additional issue in Home Assistant is the `action_command_topic`, which is what the current status is  IE: `Heating` or `Idle`.
The action_command_topic expects `heating`, `cooling`, `idle` or `off `as the dataset.  Currently your "state" attribute only sends `heating` or `off`.

Does that make sense?

austwhite, 2020-04-25:
If it helps, the documentation for Home Assistant HVAC MQTT is here :
https://www.home-assistant.io/integrations/climate.mqtt/

fashberg, 2020-04-25:
Hey!

I'ved released my  version 1.10-fas, see https://github.com/fashberg/WThermostatBeca/releases

You can find an interim version "WThermostat_1.10-fas-minimal.bin" for upgrading from Klaus Version 1.0[-5] to 1.10-fas-minimal. Then flash normal version "WThermostat_1.10-fas.bin"

@klausahrenberg  - here is how to reduce your size:
You are not defining flash structure in plaformio.ini.
I define:
```board_build.ldscript = eagle.flash.1m.ld```
Default is 'eagle.flash.1m256.ld' which adds 256kByte spiffs memory but this is not used/required. That's why there is only 385kByte available sketch memory on your 1.05 and my 1.10-fas has 640KByte available.
Your old eclipse-slober 1.00 builds are not using spiffs memory, so there was lot more space available.
My 1.09-fas was running for two weaks without any problems ore decreasing memory.
Not Using the 250k storage this could be used for my improvements (hass autoconfig, etc).

Maybe we should consider to work more togther and to merge my fork back. 
What do you think?
Is single-topic really required? If we drop then the merge would be easy

Kind Regards
Folke

austwhite, 2020-04-25:
Hi @fashberg 
I just noted on your github about a little bug I found.  :)

austwhite, 2020-04-25:
@edellamaria  as the OP of this issue, the above information may be helpful and the implementation by @fashberg (Folke) may be easier to use.

klausahrenberg, 2020-04-26:
> You are not defining flash structure in plaformio.ini.
> I define:
> `board_build.ldscript = eagle.flash.1m.ld`

@fashberg Thanks for this, I was looking around for this option in Atom IDE. As much as I like the IDE, but setting build options was easier in Sloeber :-) I changed this now and now I have a lot of available sklech size after flashing - so version 1.06 will come with this change.

> Maybe we should consider to work more together and to merge my fork back.
> What do you think?

I'm highly interested in this. Our versions developed in such different directions already, but it should be possible. Maybe we should unify the WAdapter respository first. Actual I'm working on version 1.06 and already made some fixes and optimizations (not published yet) to reduce binary file size and I modified logging again. I saw that you made some changes at WDevice classes and config page handling etc. I think, it would help us both to put WAdapter on the same base first. For this, I would finish version 1.06 first and then we could check what we have to modify to implement your changes to merge.

austwhite, 2020-04-26:
@klausahrenberg and @fashberg 
As a user of the firmware, I would very much welcome you both working together as I believe you both have the same goal and with the best of both, this firmware will be absolutely amazing.
Let me know if you need any testing done.  I have a dev environment for Home Assistant here for testing.

fashberg, 2020-04-26:
> @fashberg I'm really appreciate for you hard work on this fork.
> I have only two questions:
> 
> 1. Could you put back the restart/reboot button to the web interface? Sometimes it's needed. I know, I can do it on the http://<thermostat_ip>/reset, but would be much easier on the web interface.

Hi @iMike78 !
Thank you for the hin. The statpagebuffer was limited to 2048 (inlcluding head, style and scripts) and it got cut after the firmware upgrade button.

Fixed in https://github.com/fashberg/WThermostatBeca/releases/tag/v1.11-fas

> 2. Is there any chance to show the uptime on the info page?

Sure, it's already added some releases ago.

Kind Regards
Folke

iMike78, 2020-04-26:
> Fixed in https://github.com/fashberg/WThermostatBeca/releases/tag/v1.11-fas
Hi @fashberg!
WOW! It was so quick ;) Thank you. 
In the meantime turned out I have one more question :)
Can I set static IP somehow? It's done in the router now, but all my other nodes can be set individually (esphome nodes before compile, tasmotas are on console). Or in this case I will have to compile for myself, and write it in the code?

austwhite, 2020-04-26:
To be honest, it is better to use a reserved IP in the router than set static on the device.
It's more reliable and less likely to have IP conflicts.
That said, It could be done I am sure 🙂

iMike78, 2020-04-26:
@austwhite 
You could be right, but the problem is that, most of the routers has limited number of reserved IP slot ;) I have more than 50 appliances on my network already, most of them needs to have static IP

austwhite, 2020-04-26:
@iMike78 
Might be time for a new router :)
If you are going over 50 devices on WiFi it probably is getting close to time to look for something more robust.  
It's not hard to add a static IP to the code though.  I don't have a lot of time but @fashberg is very responsive to new features.

iMike78, 2020-04-26:
@austwhite 
I just bought a Ubiquiti EdgeRouter lite, and a UAP-AC on this week, so I'm in a big change already. And, of course, I'm in big trouble with the settings :D

austwhite, 2020-04-26:
@fashberg 
I have been using 1.11 and it is working great.  The Home Assistant discovery works great and the little bug with the manual temperature setting seems to be resolved.

austwhite, 2020-04-26:
> @austwhite
> I just bought a Ubiquiti EdgeRouter lite, and a UAP-AC on this week, so I'm in a big change already. And, of course, I'm in big trouble with the settings :D

That is some great equipment.
I suggest you check out The Hookup on YouTube.
Rob did some really great videos on the Unifi gear a little while ago

iMike78, 2020-04-26:
@austwhite 
Thanks for the suggestion, I'll check that. I know him, I'm one of his follower :D

klausahrenberg, 2020-05-19:
I will close this issue. Flash size problem is solved. In general, I recommend folowing:
- If you use Mozilla Webthings: use my firmware
- If you use Home Assistant: use the fork from @fashberg 
- If you use only MQTT: take the one, you more like :-)

closed 2020-05-19 by klausahrenberg