klausahrenberg/WThermostatBeca #100: Should I proceed with 2.22.03.00712

IanAdd, 2020-05-07:
Hi

Just got my bht-002GB-B branded Minco Heat with sticker on the box saying BHT-002GBLwifi. The board says bac-002-wifi-2019-0409.

I want to get the floor temp which is not available via the HA tuya integration so your MQTT option is what I want.

I am nervous as the version discussed in another issue thread referred to versions which sound newer. 

I don't want to brick it, what do you think?

IanAdd, 2020-05-09:
So far I've tried reading it and get either timed out waiting for packet header or invalid head of packet (0x00).
I've tried a range of baud rates.
Any thoughts?

austwhite, 2020-05-09:
I would not do the soldering method and update it using Tuya Convert.
Just be aware you may brick the device if it is not compatible.
Tuya Convert will backup the original firmware, but you would need to be able to upload it if this firmware doesn't work

klausahrenberg, 2020-05-10:
I agree, try with Tyua-convert, if possible. If not and you want to try the soldering way, please follow my Flashing instruction. The first thing is to make a backup of the original firmware. This backup is without risk and if successful you know, that the flashing of new firmware is possible. So the risk for bricking the device is quite low, if the backup worked well.

klausahrenberg, 2020-05-10:
The timout means, something with your connection is  or ESP is not in programming mode. Remove the touch unit from the relay part for flashing. Power the ESP for flashing only over  USB programmer. Connect GPIO0 to GND at power up the programmer to get the ESP in program mode. Try the backup first. If the backup runs well, the flashing should be possible.

IanAdd, 2020-05-11:
Removing the display, That's worth a try. 

I explored that gently and found it a bit resistant to coming apart. Is there a clue on how to do it?

IanAdd, 2020-05-12:
Ok, the display is soldered to the board on this model, it's not on a connector. So tuya-convert is the next plan.

closed 2020-05-19 by klausahrenberg