klausahrenberg/WThermostatBeca #102: State on AC power restore

adrez99, 2020-05-08:
I've seen that on AC power restore the thermostat always goes on, it doesn't matter wether if it was on or off before the power loss. I don't know if the default firmware works like this, but I think this is an incorrect behaviour; it should either restore previous state or be off
I'm currently on 1.03b (the fork from fashberg, because it is Home Assistant friendly) and can't test latest version, but I suppose this behaviour hasn't changed.

austwhite, 2020-05-08:
I've noticed this too.
Default firmware does not do this.  Default firmware sets to off on reboot/power failure.
This would preferred, for power in to default to off, from my view as it would be safer.

klausahrenberg, 2020-05-10:
Ok, noted for next version. MCU switches device always on after power supply. I didn't changed the power state after power on. But it's not a big thing to store the last state and restore it at next boot. Thanks for hint.

klausahrenberg, 2020-05-14:
Hi, I checked the behaviour on my model (BHT-002-GBLW) and the deviceOn state is stored at my model. If the thermostat was on before power loss, it goes on after - and if it was off before, the unit is staying off after power loss. So you wrote that you use the fork from @fashberg, maybe you should address this issue there. Maybe HA is the reason or a retained MQTT message.

austwhite, 2020-05-14:
If you are using version 1.03b. It might be worth upgrading
@fashberg is up to version 1.12 now and a lot of things are fixed, and it has Home Assistant Auto Discovery to fix any HA issues

fashberg, 2020-05-14:
Because it looks HA or fashberg-fork specific i have created an issue here https://github.com/fashberg/WThermostatBeca/issues/14

closed 2020-05-19 by klausahrenberg