klausahrenberg/WThermostatBeca #103: BHT-002 Temporary overide

markbooker, 2020-05-11:
Klaus,

I am happy I stumbled across your implementation and have got my BHT-002GC working with homeassistant using it.

Is there a way to set the manual override to revert back to auto after the next scheduled event (like traditional mechanical Thermostats do).

Regards

Mark

austwhite, 2020-05-11:
I believe this is a limitation of the thermostat from factory.
The factory firmware also does not do this.

markbooker, 2020-05-11:
Agreed, it also seems to be the same on a number of smart thermostats.

I was hoping, more praying that it could be coded in.

Regards

Mark

fashberg, 2020-05-12:
Hey! 
The Main-CPU (MCU) of the thermostat is not able to do this. The Code of the main CPU can't be changed, only the WLAN chip gets replaced by this software.

It would be possible to do this in the WLAN Chip.
But then changing to Manual Mode would switch back every time. Then there is no manual mode after adding that change.

You can easily add such logic in homeassisant using the GUI in Home Assistant.
[image]

If the thermostat switches from auto to heat for more than 20 seconds add a delay of two hours and then switch back to auto-mode.

The same as code for automations.yaml:
```yaml
- id: '1589271260132'
  alias: BecaBueroSwitchBackToSchedule
  description: SwitchClimateBackToSchedulerModeAfter2Hours
  trigger:
  - entity_id: climate.thermostat_buero
    for: 0:00:20
    from: auto
    platform: state
    to: heat
  condition: []
  action:
  - delay: '2:00:00'
  - data:
      hvac_mode: auto
    entity_id: climate.thermostat_buero
    service: climate.set_hvac_mode
```

Note: This snipped or above setting works only with my fork, the original version does not have the combined parameter "mode" with values "off", "auto", "heat".

fashberg, 2020-05-12:
Probably another simple solution:

Switch all Beacas to "auto" (regardless of current state) at 1 AM with this code:

```yaml
- id: '1589274177037'
  alias: AllBecasToAutoInTheMorning
  description: ''
  trigger:
  - at: 01:00:00
    platform: time
  condition: []
  action:
  - data:
      hvac_mode: auto
    entity_id: all
    service: climate.set_hvac_mode
```

iMike78, 2020-05-12:
> Probably another simple solution:
>
> Switch all Beacas to "auto" (regardless of current state) at 1 AM with this code:
>
> ```yaml
> - id: '1589274177037'
>   alias: AllBecasToAutoInTheMorning
>   description: [...]

I did the same timed switching before through mqtt publish, but for some reason it doesn't work anymore with the 1.11-fas fw. 
I like both of your automation, and I'll use them ;) 
Thanks again :)

markbooker, 2020-05-12:
Thanks for all the suggestions, I am happy to do the automation via
homeassistant but i was also looking for a method of diong it in the stat
software as I need to replace my parents old thermostat and I really like
this one and it is very cost efficient.

Switching back from manual to auto each time an event passes is fine for
me, as normally you only want to override for a short period of time
normally.

If you have any pointers where in the code I might make the
chanes tha would save me a lot of time going through the code.

Many thanks

Mark

fashberg, 2020-05-12:
Hey @markbooker 
> If you have any pointers where in the code I might make the chanes tha would save me a lot of time going through the code.

The real scheduling takes Place in MCU.

I would start at loop() in WBecaDevice.h:
https://github.com/fashberg/WThermostatBeca/blob/master/WThermostat/WBecaDevice.h#L434

See also updateTargetTemperature at https://github.com/fashberg/WThermostatBeca/blob/master/WThermostat/WBecaDevice.h#L1411

It takes current day and time, gets current schedule and takes the temperature (to be able to return the current targetTemperature).

I would store this->oldSchedule and check if the schedule has recently changed.
if schedulesMode->equalsString(SCHEDULES_MODE_OFF) and oldSchedule!=newSchedule and OptionSchedulerDisablesManualMode==true then enable Scheduler.

fashberg, 2020-05-12:
Hey @iMike78 
> I did the same timed switching before through mqtt publish, but for some reason it doesn't work anymore with the 1.11-fas fw.
> I like both of your automation, and I'll use them ;)
> Thanks again :)

Please open a new issue at my fork shoing some logs and the automation-code and i'm sure we can fix it together. Try  1.12-fas, which has fixed an hass autodiscovery issue. Running older versions the hass-state of the beca-devices was reset on every mqtt autodiscover-message so maybe automation got broken during unknown-state.

iMike78, 2020-05-12:
> Hey @iMike78
>
>
> Please open a new issue at my fork shoing some logs and the automation-code and i'm sure we can fix it together. Try 1.12-fas, which has fixed an hass autodiscovery issue. Running [...]

That's not important anymore for me, as I'll use your automation.
Anyway it was interesting, because when I sent an mqtt message from HA using the mqtt publish service, that's works flawlessly, but the same settings (topic and payload) in automation never worked. Unfortunately I can't send the settings, I deleted them already, and actually to set the hvac_mode is much more straightforward. :) So thanks again ;)

markbooker, 2020-05-12:
Great thanks, I'll take a look and thankyou for the super fast response to
messages.

Regards

Mark

klausahrenberg, 2020-05-18:
> Is there a way to set the manual override to revert back to auto after the next scheduled event (like traditional mechanical Thermostats do).

Hi, I have implemented this function in version 1.09. It's optional and enabled by default. Note: Device must be reconfigured after upgrade.

closed 2020-05-19 by klausahrenberg

privatesam, 2020-05-20:
> 
> 
> Klaus,
> 
> I am happy I stumbled across your implementation and have got my BHT-002GC working with homeassistant using it.
> 
> Is there a way to set the manual override to revert back to auto after the next scheduled event (like traditional mechanical Thermostats do).
> 
> Regards
> 
> Mark

Would you be happy to share you Home Assistant integration?

fashberg, 2020-05-20:
Hey @privatesam, 
Klaus is primarily using Mozilla Webthings. 
My fork at https://github.com/fashberg/WThermostatBeca has HASS Autodiscovery and also you can find some more HASS-Specific documentation.
Be aware that the fas-versions have some hass-specific mqtt improvements so not all documentation can be used at klaus version.

Kind Regards
Folke

klausahrenberg, 2020-05-20:
> Would you be happy to share you Home Assistant integration?

Hi @privatesam , take a look at the fork from @fashberg , he modified the firmware for better HA integration.

privatesam, 2020-05-21:
Thank you both @klausahrenberg and @fashberg for all your work on this - so useful!

markbooker, 2020-05-21:
@privatesam, if you still want me to share my HA setup using this setup I
am more than happy to do so

austwhite, 2020-05-21:
> @privatesam, if you still want me to share my HA setup using this setup I am more than happy to do so
> […](#)

Also be aware the @Fashberg version does HA Auto Discovery if you prefer to use that.
Just have MosQuiTTo integration setup for discovery and it just works.
Manual config is easy too if you prefer that :)