klausahrenberg/WThermostatBeca #104: BHT-002-GALW flapping between available and unavailable

tmp0r, 2020-05-13:
Hi Klaus,

thank you very much for your great work! Currently I'm testing your firmware but running into problems. My flashed BHT-002-GALW with firmware version 1.08 is flapping every ~10 seconds. When I ping the thermostat it will answer for 10 seconds, than for another 10 seconds nothing and than for 10 seconds available again:
```
root@hal01:/etc/openhab2# ping 172.16.40.101
PING 172.16.40.101 (172.16.40.101) 56(84) bytes of data.
64 bytes from 172.16.40.101: icmp_seq=2 ttl=255 time=6.78 ms
[...] 8 successful pings
64 bytes from 172.16.40.101: icmp_seq=11 ttl=255 time=5.19 ms
64 bytes from 172.16.40.101: icmp_seq=23 ttl=255 time=2.51 ms
[...] 8 successful pings
64 bytes from 172.16.40.101: icmp_seq=32 ttl=255 time=1.73 ms
```

hal01 is my smart home host which is one the same subnet as the thermostat. While this thermostat is flapping an other thermostat (BHT-002-GALW but with original Tuya firmware) is available without outages on the same network connected to the same WiFi AP and the same SSID.

The thermostat was flashed by using [tuya-convert](https://github.com/ct-Open-Source/tuya-convert). MQTT is enabled.

Please help me!

Thanks in advance and best regards,
Daniel

fashberg, 2020-05-13:
Hey Daniel!

Do you have more than one access point? (Roaming/reconnect issue)
Please check also if the device resets/crashes by checking the uptime (webpage of device->Info->Uptime)

Folke

tmp0r, 2020-05-13:
Hi Folke,

thanks for your reply. Yes, I have more than one AP but only one is active. The other one is powered off.

I took a look in my WiFi controller log:
```
Office_Thermostat has connected to wifi01-ap-og with SSID "FuIot" on channel 1 | 8:26 pm | 05/13/2020
Office_Thermostat disconnected from "FuIot" (0m connected, 4.82 MB, last wifi01-ap-og) | 8:26 pm | 05/13/2020
Office_Thermostat has connected to wifi01-ap-og with SSID "FuIot" on channel 1 | 8:26 pm | 05/13/2020
Office_Thermostat disconnected from "FuIot" (2m connected, 4.78 MB, last wifi01-ap-og) | 8:26 pm | 05/13/2020
```
It looks like the thermostat diconnects itself from the AP. All other thermostats with Tuya FW don't appear in the log.

The uptime counter wasn't reseted, still >300 minutes. Also did an reboot but without success.

Thanks again,
Daniel

austwhite, 2020-05-13:
@tmp0r 
Does your router indicate the WiFi strength of the Thermostat (often labelled as RSSI)?

tmp0r, 2020-05-13:
Hi austwhite,

the device has a signal strength of 74.3% (-60dBm).

Best regards,
Daniel

tmp0r, 2020-05-14:
Hi at all,

today I've flashed another thermostat and it works without problems and without connection issues. So I decided to do an firmware update on the first thermostat. I've uploaded v1.08 again but the thermostat is still flapping.

Best regards,
Daniel

fashberg, 2020-05-14:
You've said, that with original-firmware there was no flapping.

There are lot of other ESP-Firmwares, for example try to flash tasmota and see of WLAN works stable here https://github.com/arendst/Tasmota/releases .
Tasmota has tuya support but it is much more complicated to cotrol the thermostate.

Or try my fork of this project (https://github.com/fashberg/WThermostatBeca)
I have changed reconnect-behaviour in Wifi-Code, so maybe it's more suitable for your needs (but a reconnect should not occur). Maybe there is Band-Steering (move from 2,4 to 5GHz wifi, which will not work, because beca only supports 2,4 GHz, or channel-steering to another access point, etc..)

tmp0r, 2020-05-15:
Hi everybody,

@fashberg I've also tried your firmware but the flapping doesn't stopped.

At this point I've removed power from the thermostat so the device did an reboot and now everything works :-) No flapping, no connection issues.

Thanks to all who helped me. I'll close the case.

Best regards,
Daniel

closed 2020-05-15 by tmp0r