klausahrenberg/WThermostatBeca #105: "Not Enough Space" when trying to update to 1.10 on BHT-002-GALW

alexdtag, 2020-05-19:
My themostat BHT-002-GALW is currently on Rev 1.03
After uploading WThermostat_1.10.bin via Config -> Upload new firmware, I get the message

> Not Enough Space
> ESP reboots now...

and the device remains on 1.03.
How can I update the fw to 1.10?

fashberg, 2020-05-19:
Hey,

try to upgrade first to Version 1.07.

You can download it from here: https://github.com/klausahrenberg/WThermostatBeca/tree/053fcaccc1fc046555634a7f616470d5fb8714ff

Then go to 1.10.

If 1.07 does not fit, try another minimalized intermediate version. 

For example use wthermostat-minimal-1.12-fas.bin from here: https://github.com/fashberg/WThermostatBeca/releases (No MQTT, No WebThings, No HomeAssistant Autodiscovery) - then use fully blown version

Kind Regarsds
Folke

alexdtag, 2020-05-19:
Thanks a lot, @fashberg - the update to 1.07 worked, and then to 1.10 was successful, too 👍 

Thanks!
Alex

closed 2020-05-19 by alexdtag