klausahrenberg/WThermostatBeca #106: Settings reset

tshcherban, 2020-05-20:
How can i reset thermostat to initial configuration? Previously configured wifi is not available, so im unable to access its web ui.

fashberg, 2020-05-20:
The official way is 
* Power off the device by Pressing the button in the middle.
* Then press the "down" button (most right) for about 8 seconds.
* The Screen starts to blink and shows a WiFi-Icon.

Then the MCU (Main-CPU) requests a Wifi-Reset from Wifi-CPU via the internal serial connection.

I've implemented in my fork, which is currently not merged with this version, a reset of wifi settings if you do this key sequence. 
 https://github.com/fashberg/WThermostatBeca/blob/master/Configuration.md#network-recovery

Using the original version the blinking will start but the settings will not be switched to defaults.

If you have used tuya convert for initially flashing: This process is only possible with original tuya-cloud-firmware. 

So it seems neccessary to solder the required pins to your ESP as described in Flashing.md

If you have misspelled  WIFI Name or PreSharedKey maybe you can take a look to your browsers history.
Maybe you can find something like http://IpAddressOfThermostat/saveConfiguration?i=thermostat-flur&s=LUFTFUNK&p=___NOCHANGE___

If you can find it you can see SSID and PSK in Plain-Text (__NOCHANGE__ is not my PSK :) )
Then reconfigure your Wifi-AP temporarily to the "wrong" settings and you should be able to access your Thermostat

This worked once for me. Then i've imlemented to wifi-reset feature. All my thermostates are converted with tuya.

Kind Regards and Good Luck!

Folke

tshcherban, 2020-05-20:
Many thanks! Ill try your fork.
Fortuantely i haven't forgot or misstyped a password or SSID, just curious for the future.
Can your firmware be uploaded to thermostat via 'Update firmware' btn? And what else have you changed/added in your fork?

klausahrenberg, 2020-05-21:
Sorry for the inconvenience. Yes, it's a bug in my version. Once you typed false credentials for Wifi, you can't correct this and also reflashing by serial does not reset wrong settings. 
Solution is to flash any other firmware (e.g. @fashberg). After that you can reflash my firmware and reconfigure network settings. That answers also your second question. You can use upload firmware button to switch between the different firmwares. You could also upload Tasmota. 
I will correct this bug in my firmware in the way that @fashberg has it modified.

fashberg, 2020-05-21:
Hi @tshcherban ,

> Can your firmware be uploaded to the thermostat via 'Update firmware' btn?
Yes, thats possible. You can also flash tasmota or even you can flash the original firmware back.

After flashing you have to reconfigure the device (Thermostat is AP with default settings).
For some klaus-versions i keep Wifi Settings.

> And what else have you changed/added in your fork?
Primarily better HASS support, see https://github.com/fashberg/WThermostatBeca/blob/master/README.md

> P.S. Are all theese thermostats equipped with relay? Or electric floor unit have thyristor? Relay is quite noisy when changing state, im afraid it might be a problem at night time.

See https://github.com/fashberg/WThermostatBeca/blob/master/README.md#hardware-versions 
I don't think that 16 Ampere Electical Heating Version has Thyristor.
The layout of the PCB track shows several Options for the three versions GA, GB or GC. But all of them seem to be relay-based. 

My GA-Version has two relays. One for Open, one for Close.
https://github.com/fashberg/WThermostatBeca/blob/master/docs/install-003.jpg

I only need to "OPEN" relay. If heating starts, then first CLOSE relay goes off, then OPEN relays goes ON. You have the sound two times, GB and GC version have only one relay.

But the Becas are not that noisy than my old devices with bimetal !
So i'm happy.
And: it's getting more silent if installed in the wall!

Kind Regards

Folke

klausahrenberg, 2020-05-28:
Hi, I checked and optimized my firmware regarding this issue. If you saved a wrong wifi configuration, you can press the 'down'-button, if the thermostat is off for 8 seconds until the screen blinks. If the thermostat is not connected to wifi (because settings are wrong for example), the device will create a Wifi-Hotspot. Try latest version 1.11. If you still have troubles, let me know - please also tell tested version.

closed 2020-05-28 by klausahrenberg