klausahrenberg/WThermostatBeca #109: New device? ... BHT-002-GALW (warm water floor heating)

mf01, 2020-05-23:
Currently these devices are listed within the select-box

Floor heating (BHT-002-GBLW)
Heating, Cooling, Ventilation (BAC-002-ALW)
Floor heating (ET-81W)
Floor heating (Floureon HY08WE)

I'm using BHT-002-GALW (warm water floor heating)

Firmware 1.10 is working and reports this:

`thermostatbeca_1/thermostat/properties {"idx":"thermostatbeca_1","ip":"x.x.x.x","firmware":"1.10","temperature":26.00,"targetTemperature":15.00,"deviceOn":true,"schedulesMode":"off","ecoMode":false,"locked":false,"floorTemperature":0.00,"state":"heating"}`

I'm a little bit confused about the "state: heating" even targetTemperature is lower than current temp.

I think GBLW and GALW are working in a different way. It's possible to add this device, too?

klausahrenberg, 2020-05-24:
Both device should work with the firmware.

Hi, parameter 'state' needs a hardware modification. See description here: https://github.com/klausahrenberg/WThermostatBeca/blob/master/Configuration.md#1-configure-thermostat-device-model-selection. If hw modification done, it shows the state of the relay. Otherwise, please deactivate this option and the parameter 'state' will not appear anymore.

mf01, 2020-05-24:
Thx. Have added 2x "classic" R=10k. Need to order SMD resistors to can close the case and power with 230V instead of 3.3V the ESP only.

Current output...
`{"idx":"beca-az","ip":"10.6.0.97","firmware":"1.10","temperature":25.50,"targetTemperature":35.00,"deviceOn":true,"schedulesMode":"off","ecoMode":false,"locked":false,"floorTemperature":127.00,"state":"off"}`

After receiving SMD elements I will update the status.

mf01, 2020-06-06:
Have added 2x R=10k SMD1206 as described in the link above. It works as expected. And "heating" is only shown if target is higher than current temperature!

`{"idx":"beca","ip":"x.x.x.x","firmware":"1.10","temperature":23.00,"targetTemperature":22.00,"deviceOn":true,"schedulesMode":"off","ecoMode":false,"locked":false,"floorTemperature":0.00,"state":"off"}`

_First have ordered SMD0603. But this was too small for me. :-D_

closed 2020-09-23 by klausahrenberg