klausahrenberg/WThermostatBeca #112: Update from 1.05 to 1.11 the easy way? *running out of memory*

ABehr87, 2020-06-02:
Hello,

I converted my bht-002-gblw with tuya convert as 1.05 Version was up to date. But now I can´t update it because the system runs out of memory if I try to upload the current Firmware.
Do I need to reflash it or is there a workaround to do it easier?
Thank you very much!

Best wishes,
Andreas

austwhite, 2020-06-02:
Use @Fashberg minimal install first then update.
Note, you will lose settings when you do this

klausahrenberg, 2020-06-02:
Yepp, I used wrong compiler settings in 1.05. As @austwhite wrote, use the minmal version from @fashberg. After that you can flash the new 1.11. I corrected the compiler settings now - so this is a single issue and will not exist at newer versions

closed 2020-06-03 by ABehr87