klausahrenberg/WThermostatBeca #113: Can't receive JSON for ''<topic>/thermostat/properties"

dima11235, 2020-06-06:
Hello Klausahrenberg,

I am setting up an Beca thermostat updated with your 1.11 firmware in Home Assistant with Mosquitto brocker.

Receiving one specified variable by MQTT works well, for example I can successfully receive value for  <topic>/thermostat/properties/deviceOn (returns true or false).

Also I can successfully read value for <topic>/clock/properties which returns JSON like {"ntpServer":"pool.ntp.org","timeZoneServer":"http://worldtimeapi.org/api/ip","epochTime":1591484152,"epochTimeFormatted":"2020-06-06 22:55:52","validTime":true}. 

But if I try to read <topic>/thermostat/properties, it returns "unknown".

Is it possible to receive JSON value for <topic>/thermostat/properties or this works only for <topic>/clock/properties?

fashberg, 2020-06-06:
Hey,

Try my HASS autodiscover  fork from https://github.com/fashberg/WThermostatBeca

Kind regards

Folke

fashberg, 2020-06-07:
Update:
If you send empty string
```mosquitto_pub -h mqtt -m "" -t "home/buero/thermostat/state/deviceOn"```
you'll receive two lines
```
home/buero/thermostat/state/deviceOn (null)
home/buero/thermostat/state/deviceOn true
```

The first is your request, the second the response

You can do the same with all (json) properties:
Request:
```mosquitto_pub -h mqtt -m "" -t "home/buero/thermostat/state"```
Reponses:
```home/buero/thermostat/state (null)
home/buero/thermostat/state {"idx":"therm-buero","ip":"10.10.200.46","firmware":"1.11","temperature":23.00,"targetTemperature":10.00,"deviceOn":true,"schedulesMode":"auto","ecoMode":false,"locked":false,"floorTemperature":0.00,"state":"heating"}```

klausahrenberg, 2020-06-09:
Hi,
If you don't receive the complete state at once, please check in the thermostat device configuration, if option 'Send changes in separate MQTT messages' is selected.
If this option is selected, then the device sends never a complete json with all values, it will sent all parameters by a single mqtt message with topic e.g.  ```topic/thermostat/properties/deviceOn```.
If you send an empty payload to ```topic/thermostat/properties```, it will answer with a message for every parameter with the specific topic.
So to answer your question:

> Is it possible to receive JSON value for topic/thermostat/properties or this works only for topic/clock/properties?

Yes it's possible, please deactivate the option 'Send changes in separate MQTT messages'. Then the behaviour will be exactly the same like for the clock.

dima11235, 2020-06-09:
Hello Fashberg, Klausahrenberg,

Thank you very much for your comments and great custom FWs for Beca thermostats!

Currently I installed the Fashberg's FW. My thermostats were successfully integrated into Home Assistant environment with MQTT auto discovery.

closed 2020-06-09 by dima11235