klausahrenberg/WThermostatBeca #114: Support for BME280 possible?

mf01, 2020-06-06:
I'm looking since a long time for a wall thermostat which can also show humidity (and pressure).

Unfortunately I didn't found something. Currently I'm using BHT-002 (GALW). 

I'm thinking that's can be possible to solder a BME280 module directly to the ESP. Some space should be available in the box.

Because of GPIO5 is used for detecting the relay status it possible to move this to another pin?

GPIO4 and GPIO5 are used for I2C.

> GPIO 4 is SDA
> GPIO 5 is SCL

Or maybe it's possible to user another unused pins on the ESP for this?

If possible the thermostat can also show humidity (and maybe pressure) at the display? Otherwise only MQTT is supported. Thought about toggleling the current temperature value at the display.

Sorry for this stupid question... but maybe an enhancement?

klausahrenberg, 2020-06-09:
Hi,
interesting idea. 

> f possible the thermostat can also show humidity (and maybe pressure) at the display?

Sorry, no way to do that on the display - it would only be possible additionally via MQTT.

However, to connect the additional sensor for humidity and pressure might be a good idea. But how should it work with temperature? You can not use the more precise temperature value of external sensor with the thermostat. There is no way to override the values of thermostat sensor and use it to control the relay. To avoid confusion with 2 different temperature values I could only implement the additional humidity and pressure values -  which is kind of sad.

mf01, 2020-06-09:
> no way to do that on the display

Yeah. I thought this. Because WT is a closed system I think. The ESP is only for remote controlling.
But the use of the additional sensor would be even nice. In this case it not needed to use additional sensors, like battery powered Aqara Zigbee sensors.

> it would only be possible additionally via MQTT

If possible that would be a good enhancement. But the main question is .. It possible to use GPIO4 + GPIO5 for I2C. Relay detection need to move to another GPIO. And should be configured by UI to keep current hardware modification usable furthermore.

UlrichMai, 2020-09-02:
hi @mf-1, how is your BME280 project progressing ? I just ordered 3 of those to try out.

UlrichMai, 2020-09-08:
[image]
This how I placed the sensor on the board. I used double-sided adhesive tape 1mm thick besides the sensor, so the sensor is still above the pcb.

mf-1, 2020-09-10:
@UlrichMai Nice pic... I need to order some new WT's, too. Then I want to try with these, too. Currently I've already 5 WT's in action without BME280.

Regarding your question about "my" status ... The plan was to order some new WT's. See above. The part of firmware modification would be "the" challenge because I've never created a C++ program to use as firmware. How did you solved this?

Btw. your black wire is a little bit hidden in you pic. :-D

UlrichMai, 2020-09-10:
@mf-1, the black wire connects to GND, that is the upper right corner of the chip. I will shot some better pictures when I modify the next.
Working with the sensors is very easy (for a programmer :-)
```c
#include <Adafruit_BME280.h>

in setup():
Adafruit_BME280 bme;
bme.begin():

in loop() every 10 sec
humidity = bme.readHumidity();
```

My firmware is still in development, it is based on [MikesThermostat](https://github.com/bikemike/MikesThermostat). I am currently targeting HomeKit, may be later MQTT.

UlrichMai, 2020-09-11:
@mf-1,  I use breadboard cables too. They are kind off thick for this situation. Any recommendation for thinner still flexible wire?

UlrichMai, 2020-09-11:
@mf-1, I bought that thermostat 2 weeks ago: https://www.amazon.de/dp/B07GSRQJ11
I was able flash it using tuya-convert. I ordered the same again, they arrived yesterday, but tuya-convert is not able to flash them. How are you flashing the devices?

fashberg, 2020-09-11:
Hey @UlrichMai 

I have flashed 10 Thermostates with tuya without any problems.

Once i've uploaded while coding at my fork a buggy version and had to solder.
Then i've found a real great idea and 3d-printed an ESP-pogo Adapter. Documented here: 
https://github.com/fashberg/WThermostatBeca/blob/master/Flashing.md#self-made-esp-pogo-adapter

Kind Regards

mf-1, 2020-09-11:
> How are you flashing the devices?

@UlrichMai First device (order only once for testing) I've soldered and tried to flash firmware. Without success .... whatever the reason was... I don't know.

2nd device (I had need to order and wait) I've soldered wire, too. Flash was successful. I tried the the 1st one and it was also possible. I tried Tasmota first ... on both devices. Or maybe as 2nd try at 1st devices. I can't remember. But I was able to "upgrade" _WThermostatBeca_ using Tasmota upgrade tool. It was needed to switch that other firmware can be uploaded.

Than I've ordered 3 more devices and tried tuya-convert. A Pi3 was now free because of upgrading to Pi4. So WiFi was available :-D I need to try with an old WiFi AC stick to use it on my computer it-self, instead of Pi3. 

All of these 5 devices are using now the current 1.11 firmware. I need to order 3 more devices for the other room WT replacements. Then I would be start with BME280 tests outside "production", means not controlling room temperatures. :)

> I bought that thermostat 2 weeks ago: https://www.amazon.de/dp/B07GSRQJ11

I've ordered all my devices at AliExpress. 1st and 2nd each ~23€ ... 3rd to 5th each  ~18€ ....

> Then i've found a real great idea and 3d-printed an ESP-pogo Adapter. Documented here:
> https://github.com/fashberg/WThermostatBeca/blob/master/Flashing.md#self-made-esp-pogo-adapter

@fashberg Thx. I was looking for this too. I've another ESP board (no WT) which seems buggy and I cannot upload something using the board's pin header. I need to try to flash the ESP directly.

fashberg, 2020-09-11:
Hey @mf-1  and @UlrichMai,

I'ved ordered my first for testing at amazon.de and later ones at AliExpress (Moes Official)

On the tasmota website you can read for some devices: "since the template for this devices got listed at tasmota they've the firmware so that tuya-convert is not longer possible"

The easiest way using tuya-convert is on Raspi 4, connected with LAN to Home-Network and using Raspi's-WiFi for tuya-convert. Formerly i've used ubuntu from stick with permanent partion (can be easyly created with <https://rufus.ie/>)

@UlrichMai : Worst case; They've changed to WRG1 Chip, which is the new tuya-replacement for ESP, see #132 

Folke

UlrichMai, 2020-09-11:
@fashberg, the chip is still the same ESP. I see errors in the tuya-convert logs like this:
```
new client on port 443 from 10.42.42.38:32388
ID: 0242416f68626d6436614739314946523141cf6edc400c8e6022e2707d486ff087b54f31f15f
0839b198a718d5846ab152
PSK: 7debe0b3ed71d332a9e54d04ddacba80158f77e736b945b3ea7d34857e94b9fb
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] de
cryption failed or bad record mac (_ssl.c:1056)
```
that lead me to these [discussions](https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02).  I think tuya has made some changes and tuya-convert needs to catch up.
For now this mean soldering for me, 2 of 4 already done. I need to find someone with a 3d-printer :-)

fashberg, 2020-09-11:
@UlrichMai 

> I think tuya has made some changes and tuya-convert needs to catch up.

OK, abd to hear but good to know. 

> I need to find someone with a 3d-printer :-)

If you're near munich i can print you one

UlrichMai, 2020-09-13:
@fashberg, is Stuttgart close enough ;-) . I am interested in advice on 3d printers and aliexpress. Should we take this offline, can I write you an email?

fashberg, 2020-09-14:
Hey @UlrichMai 
You should have got an E-Mail to the Mailaddress i could found at your GitHub-Data (u...@o....k.com)

Folke

mf-1, 2020-09-23:
@UlrichMai @fashberg Three new devices has been arrived. :)
Try next days to solder the BME to it....

@klausahrenberg Any possibility to have somewhere in the WebUI a field to activate BME via I2C and moving relay observer to another pin?

mf-1, 2021-10-03:
@UlrichMai Today I've soldered my last/"new" three devices as your pic described ... Yeah I know... a bit late.... 
Firmware isn't updated yet. Will use Tuya-Convert next days.

Do you have an hint for changing Relay pin and get all BME data through MQTT?

UlrichMai, 2021-10-03:
@mf-1, good luck with tuya convert. It did not really work for me. Finally my solution was to build the pogo adapter that @fashberg mentioned above. I used my own [firmware ](https://github.com/UlrichMai/MaiThermostat) for a year now, but switched all my 10 devices last week to an [esphome ](https://esphome.io/) build version. This is so much easier than coding everything in C. I am using it with Home Assistant but there is not need to. You can build a firmware version that will use mqtt only. The version below should not need Home Assistant, just an mqtt broker. If you like to go the esphome way, let my know, I am happy to support. I needed to tweak the tuya climate library a bit to support the relay pin. I have not yet published the changes. Support for the BME280 is there out of the box.

Ulrich

```thermostat_mqtt.yaml
substitutions:
  th_name: thermostat0
  th_room: Labor

esphome:
  name: ${th_name}
  platform: ESP8266
  board: esp01_1m 
  platformio_options:
    upload_speed: 74880
# external_components:
#   - source:
#       type: local
#       path: my_components    

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  #use_address: ${th_ip}
logger:
  baud_rate: 0

ota:

#api:

mqtt:
  broker: 192.x.x.x   
  
time:
#  - platform: homeassistant
  - platform: sntp
    id: homeassistant_time

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

# Register the Tuya MCU connection
tuya:
  time_id: homeassistant_time
  ignore_mcu_update_on_datapoints: 102

climate:
  - platform: tuya
    name: "${th_room} Thermostat"
    id: "${th_name}_thermostat"
    switch_datapoint: 1
    target_temperature_datapoint: 2
    target_temperature_multiplier: 0.5
    current_temperature_datapoint: 3
    current_temperature_multiplier: 0.5
    # heat_state_pin: 
    #   number: GPIO13 
    #   inverted: False      
    visual:
      min_temperature: 18 °C
      max_temperature: 25 °C
      temperature_step: 0.5 °C

i2c:
  sda: GPIO4
  scl: GPIO5
  scan: true

sensor:
  - platform: bme280
    address: 0x76
    update_interval: 10s                    
    # temperature:
    #   name: "${th_room} Temperatur"
    #   oversampling: 16x
    #   filters:
    #     - offset: -3.0
    # pressure:
    #   name: "${th_room} Luftdruck"
    humidity:
      name: "${th_room} Feuchtigkeit"
      filters:
        - offset: 20.0         
```

DerDom, 2022-01-17:
Hey @klausahrenberg, it would be awesome if WThermostatBeca would support soldering a humidity sensor to the thermostats! Access via MQTT is more than enough for me: I would like to use ioBroker notify me in case humidity is too high.

The only hardware that I could find with a humidity sensor was this thermostat: https://www.amazon.de/dp/B088WGSRZH but I am not sure whether it could be flashed with WThermostatBeca and whether I could use MQTT to get humidity data. Do you happen to know?

DerDom, 2022-11-02:
@mf-1 did you get the BME280 to work? I am still very much interessted in humidity data for the WThermostatBeca...