klausahrenberg/WThermostatBeca #116: Flashing problems and tips

john4242, 2020-06-11:
This is a success story in the end, with some struggles.
The website is generally good at explaining the process:
1. Connect it all up like on the image
2. Make sure the "Blue right: ESP-GPIO0, must be connected with GND during power up" is grounded
3. Occasionally briefly grounding the "Blue left: ESP-Reset, connect to GND to restart the ESP" reset can help between read/write steps

The issue I had is that you need enough power. Neither the FT232RL module or the CH340G USB module provides enough power! They provide around 30-50 milliamps, which is not enough, so don't even try. (I did.)
The solution is to use an external 3.3V regulator module to drop the USB 5V to a stable 3.3V, and also make sure to have good USB power (first I plugged it into a cheap USB hub, which caused problems).

After these steps, things worked like a charm with my BHT-002, flash_id, read, write, all good!

fashberg, 2020-06-26:
Hey!
I will also share some of my experiences:

I've used tuya-convert for all my devices with no problems, but i've also tested flashing with both this TTL-Adapters and had success:
* DSD TECH SH-U09C2 with FT232RL
  * https://www.amazon.de/gp/product/B07TXVRQ7V/
  * 10 EUR Amazon Prime (Germany)
  * Has separate Voltage regulator on Board which delivers enough power
  * You can jumper betweeen 1,3/3,3/5 Volt (it switches Vcc AND Signal-Level - which is important)
  * Not sure if it's original FTDI-Chip (has not the 2014 'bug', but text seems to be printed not laser-engraved), but overall quality looks very good, including RX/TX LEDs
* AliExpress Cheap Adaptter with Profilic PL2303
  * https://www.aliexpress.com/item/32893637666.html
  * 3 EUR and long delivery time
  * Vcc Pin can easily soldered from 5V to 3,3V (has to be changed)
  * Signal Level is always 3,3 Volt (measured with Osci)
  * No separate Voltage Regulator, but flashing was ok

Sample Output flashing on windows with PlatformIO's python and esptool.py (so you don't have to install seperately):
https://github.com/fashberg/WThermostatBeca/blob/master/Flashing.md#option-2-flashing-device-manually

If not entering programming-mode (GPIO-0 not connected to GND) the ESP boots normally and is reachable through WiFi.

[image]

Master-Rudi, 2021-01-13:
No 3D Printer? Want a ESP-Pogo-Adapter?

--> Mail me! master.rudi@web.de

JinKanzaki, 2021-02-28:
> This is a success story in the end, with some struggles.
> The website is generally good at explaining the process:
>
> 1. Connect it all up like on the image
> 2. Make sure the "Blue right: ESP-GPIO0, [...]

THANK YOU! It reallz was the lack of power. Used my adjustable DC PSU and it worked like a charm. Dear freaking god! I wasted 4 hours of googling and soldering on this.
Fun fact: it requires almost exactly 60 mA, so that was the issue.
I was about to give up.