klausahrenberg/WThermostatBeca #118: Webthings title / use of idx possible?

mf01, 2020-06-16:
I've nothing seen to can change device title for Webthings.

I'd suggest to use `idx` as Webthing title, too.

I have some thermostats and cannot match them to the correct rooms. :)

fashberg, 2020-06-26:
Hey @mf01!

I'm not a webthings user, i've forked klaus Firmware and made some Home Assistant improvements.
But i've tested Webthings and had also that issue and fixed it in my fork some month ago.
Info: I've moved the webthings URL from root-folder to /things and adopted WebThings AutoDiscovery. Feedback welcome. In General i think klaus Firmware is better webthings tested but you can give a try :)

https://github.com/fashberg/WThermostatBeca

mf01, 2020-06-30:
@fashberg Sorry. I have only tested Webthings with few devices. And I cannot see which device is in which room. :)
It was only a suggestion to use the user input to have it also in Webthings available instead of a common name.

fashberg, 2020-06-30:
Hey,
You can test this version: [wthermostat-1.16beta-fas.bin.zip](https://github.com/klausahrenberg/WThermostatBeca/files/4853822/wthermostat-1.16beta-fas.bin.zip)

i've broke up mDNS for WebThings Autodiscovery. Now it looks like:
[image]
[image]

Kind regards
Folke

klausahrenberg, 2020-07-14:
> I'd suggest to use idx as Webthing title, too.

Yes, is used as webthing id in my firmware. I control more than one thermostat and you have to give every thermostat a different idx (and also a MQTT topic recommended). The network name of the ESP-device is taken by Webthings to detect different devices, this name is build by the idx in my firmware.

So first configure the idx and after that add it in Webthings.

Fork from @fashberg I can only recommend for Home Assistant. If you use Webthing, please use my firmware. I tested the other firmware a while ago and as fahsberg already wrote, the auto-detection is not working there, also the property names changed a lot and doesnt fit to Webthings (e.g. action heating)

closed 2020-07-14 by klausahrenberg