klausahrenberg/WThermostatBeca #119: Non WiFi Versions

Sthopeless, 2020-06-17:
Hello everyone! 
Does anyone happen to have a non wifi version of the thermostat's, I wonder if the PCB is the same without the wifi TYWE3S module? 
The price difference on some is significant and soldering a esp8266 module shouldn't be hard.

fashberg, 2020-06-17:
Hey,
interesting question. Probably some other parts, which are only required for WiFi-Mode, are not soldered and last but not least maybe the firmware in the MCU could be different with no support for serial interface to ESP.

[image]

For example the small chip HT1381 seems to be for timing a serial connections (that one between MCU and ESP).
Also the PCB is labeled with "wifi". So probably there's a non-wifi version.

Kind Regards
Folke

fashberg, 2020-06-17:
I've ordered a black BHT-002-GA(No Wifi) with china shipping for 13,40 EUR at AliExpress the answer that question. Will keep you updated.

Hint: Moes Official Store at Ali Express has selectable "Shipping From China" and "Shipping from Spain" for some variants. I've selected 'Spain', but they shipped from China through Estland and i had to pay customs (EU Import).

Sthopeless, 2020-06-17:
@fashberg thanks for your quick reply and for "biting the bullet"
If the board is the same without the esp8266 soldered maybe we can solder a esp03 or even a esp12.
Thanks again and keep us updated 👍

fashberg, 2020-06-30:
Hey @Sthopeless ,

14 days and 15 EUR later here is the answer: https://github.com/fashberg/WThermostatBeca/blob/master/Flashing.md#1-check-your-device 

Seems to be impossible!

Kind Regards
Folke

Sthopeless, 2020-07-01:
Hey @fashberg!

That's very unfortunate. Doesn't look good :disappointed: 

But I see all main MCU's are the same on both versions, do you think they are running the same firmware?!
What if RX and TX path's are traced and you try to connect a nodemcu and see if there's any console activity?!

Anyway's thank you so much! :) 

Regards,
Sthope

fashberg, 2020-07-01:
Hey @Sthopeless 

OK, going deeper. Result: https://github.com/fashberg/WThermostatBeca/blob/master/docs/Diagnose%20Can%20we%20add%20WiFi%20to%20Non-Wifi%20Thermostat%20bac-002.pdf

Kind Regards

closed 2020-07-14 by klausahrenberg