klausahrenberg/WThermostatBeca #120: Firmware 1.11: Schedules mode turns on automatically

ReachyBayern, 2020-06-17:
Hi, with version WThermostat_1.11.bin I have the situation that the schedules mode will be turned into automatic mode automatically. With  version 1.02 it turns into "on" automatically. I don't use this feature.  But if I switch the schedule mode off some hours later it is on again. What I found out is, that it turns on according the times which are there out of the firmware update... could you please check this? Or should I use the older version? Or is this a new feature which I don't Understand? 

Kind Regards 

Reachy

ReachyBayern, 2020-06-17:
Found a comment that it can be deactivated in Thermostat - Auto Mode

closed 2020-06-17 by ReachyBayern

tibo888, 2020-09-26:
Hello,
I have the same problem, can you explain how you fix it?
thanks

ReachyBayern, 2020-09-27:
Go into configuration to the item "configure thermostat" and deactivate the check box for "auto mode after manual mode...."