klausahrenberg/WThermostatBeca #121: Can ping but can't connect or receive messages

farfenkugell, 2020-06-20:
Hello,

I have BHT-002-GBLW, at first flashed tasmota, then on top of it, via it;s menu your .bin.
Connected to an access point via phone, configured wifi and MQTT, selected correct model.

Now I can see it appearing in the correct network, I can ping it, but that is pretty much it.
I cannot access it via browser nor MQTT reports anything.

Any suggestions?

fashberg, 2020-06-20:
Hey,

Can you access the webpage (http://192.168.4.1/config) in Recovery mode from your smartphone?

Also try to power cycle the device (not just pressing the Powerbutton).

Kind Regards

Folke

farfenkugell, 2020-06-20:
I can access it after holding down button for 8s, it does take a while to start an access point and sometimes pages do not load, or require a refresh. Firmware update page sends an empty response or timeouts. 

I did power cycle it down multiple times

fashberg, 2020-06-21:
You mean fw update page does not load even in Recovery mode?

And it's not a wifi signal strength issue in normal Mode? (Your Smartphone is next the thermostate and the router probably too far away).
Is the ping ok? (Timeouts, RTT)

Which firmware are you using? 1.11. 
Was everything before fine using tasmota or the original fw?

Kind regards

fashberg, 2020-06-21:
Firmware 1.11 is according to screenshot.

fashberg, 2020-06-21:
I've flashed Tasmota 8.3.1 to a test device (Moes BHT-002-GBLW). 

Failed to flash WThermostat directly ("not compatible"), found "SetOption78" which has to be set to 1 (to disable Version check on Tasmota upgrade).

Then i was able to flash klaus version 1.11 and everything was fine. Than flashed back to my fork. Everything was fine.

Broken device? Broken wifi? Broken Flash? Buggy MQTT Server which causes to crash WThermostat? Is there a boot-cycle (check uptime at info page). 
Are you able to flash a new firmware (in recovery mode)?

farfenkugell, 2020-06-21:
In non-recovery mode I cannot access it at all in any way, only ping it.
Wifi signal should not be an issue. 
There are other 3 ESP8266 devices in the  apartment, all of them are further, one of them even in same room, but further corner. 
https://i.imgur.com/gAeKga7.png

Firmware is 1.11 and I had tasmota for like 5minutes until clashed it after SetOption78, original firmware was working, but reporting x2 temperature in HomeAssistand (known bug).

Info, shows uptime 890 minutes, left it alone for messing around with it half of the day:
https://i.imgur.com/heQZbua.jpg

Firmware upgrade firstly loads almost blank page:
https://i.imgur.com/LWVgL5T.jpg
Then loads fully, wanted to downgrade:
https://i.imgur.com/OXr1OAA.jpg

But then it just times out:
https://i.imgur.com/aATTySx.jpg

Device it self is working, I can hear releay clicking, wifi is working and pings go back/forth with no dropped packages.
Regarding MQTT, I am new to those and so far have only used default unraid docker for it with my own user account.
It does not seem to crash, as uptime is running and ping working.
At the moment it does not seem I am able to flash new firmware via 192.168.4.1

farfenkugell, 2020-06-21:
I am an ape... night sleep what was I needed.
I have two esp devices in bathroom and I reset them by just flipping the switch in fuse box thus all the time I was trying to connect to my temp/humidity sensor.
While "configured" thermostat to use 5Ghz network... once fixed that I can access it now.

All what is left is to make it work with MQTT and HomeAssistant.

fashberg, 2020-06-21:
ahh - yes. Only 2,4 GHz.

For better HomeAssistant you should try my HASS-Fork at https://github.com/fashberg/WThermostatBeca

farfenkugell, 2020-06-21:
I knew it, but had brain fart, I have two access points:
so internet much slow
so internet much fast

yea.. original... after flashing your firmware all just started working with no changed needed at all. 
Already have it in Home Assistant, tested, working.
Thank you.

closed 2020-06-21 by farfenkugell