klausahrenberg/WThermostatBeca #123: does not appear in HA

madspeed, 2020-06-24:
Hi!
I have followed the instruction but thermostat does not appear in HA
What is wrong?

configuration:
mqtt:
  broker: 192.168.1.10
  port: 1883
  username: mqtt
  password: mqtt

Mqtt log:
1593020643: New client connected from 172.30.32.1 as thermostat-672472 (p2, c1, k15, u'mqtt').
1593020727: New connection from 192.168.1.10 on port 1883.
1593020727: New client connected from 192.168.1.10 as auto-62311AA4-28F7-28CC-87EC-1094AB890EFE (p2, c1, k60, u'mqtt').
1593021148: Client auto-62311AA4-28F7-28CC-87EC-1094AB890EFE disconnected.
1593021216: New connection from 192.168.1.10 on port 1883.
[INFO] found mqtt on local database
1593021217: New client connected from 192.168.1.10 as auto-D4556BE8-A396-3D39-0167-78169542EF45 (p2, c1, k60, u'mqtt').
1593022167: Client auto-D4556BE8-A396-3D39-0167-78169542EF45 disconnected.
1593022247: New connection from 192.168.1.10 on port 1883.
[INFO] found mqtt on local database
1593022248: New client connected from 192.168.1.10 as auto-382C8586-1705-1D32-4279-5ADB8CFE252F (p2, c1, k60, u'mqtt').
1593022427: Client auto-382C8586-1705-1D32-4279-5ADB8CFE252F disconnected.
1593022432: Saving in-memory database to /data/mosquitto.db.
1593022462: New connection from 192.168.1.10 on port 1883.
1593022462: New client connected from 192.168.1.10 as auto-99F3B5B8-5599-833D-F33D-2D9B2165A645 (p2, c1, k60, u'mqtt').
1593023120: Client auto-99F3B5B8-5599-833D-F33D-2D9B2165A645 disconnected.
1593023163: New connection from 192.168.1.10 on port 1883.
[INFO] found mqtt on local database
1593023164: New client connected from 192.168.1.10 as auto-E9F227E9-F5D2-B52B-9B21-64312EBD2391 (p2, c1, k60, u'mqtt').
1593024233: Saving in-memory database to /data/mosquitto.db.

fashberg, 2020-06-24:
Hey @madspeed,

if you expect HA autodiscovery you can flash my fork at https://github.com/fashberg/WThermostatBeca

Otherwise you have to configure manually HA

Kind Regards

Folke

madspeed, 2020-06-24:
Could you please tell me which file and where should I flash?
Should I flash another firmware to thermostat?

madspeed, 2020-06-24:
I have checked. I have already flashed your firmware 1.15-fas, but it does not appear in HA.
May be I should add some other settings to HA?

fashberg, 2020-06-24:
Is it an heating thermostat or climate control?

For climatecontrol you have to configure currently manually, check 
 Doc at https://github.com/fashberg/WThermostatBeca#hass-manual-configuration

madspeed, 2020-06-24:
Floor heating

fashberg, 2020-06-24:
Hmm. Seems to be more HASS specific.

Do you have autodiscover enabled in HASS?
https://github.com/fashberg/WThermostatBeca#hass-autodiscovery

And in thermostat network config you have to enable.

Are there other MQTT devices already working in your HASS environment?

Is thermostat itself working?
 
Do you see statusreport of thermostat?

```mosquitto_sub -h 192.168.1.10 -u mqtt -P mqtt -v -t "YOURTOPIC/#" ```

madspeed, 2020-06-25:
autodiscovery enabled in HA
autodiscovery enabled in Thermostat
MQTT clean installation, not used before
Thermostat is ON, connected to WifI

Where should I input it?
mosquitto_sub -h 192.168.1.10 -u mqtt -P mqtt -v -t "YOURTOPIC/#"

fashberg, 2020-06-25:
> Where should I input it?
> mosquitto_sub -h 192.168.1.10 -u mqtt -P mqtt -v -t "YOURTOPIC/#"

At your Linux CLI.
If not installed, install with ```sudo apt-get install mosquitto-clients```
Replace YOURTOPIC with the topic you have set at networkconfigpage at thermostat.
Then you should see all MQTT Traffic from and to the thermostat.
You can also run ```mosquitto_sub -h 192.168.1.10 -u mqtt -P mqtt -v -t "YOURTOPIC/#"``` to see all MQTT traffic.

If you can reach thermostat webpage , have MQTT enabled but don't see any MQTT Traffic there is something wrong with your mqtt-server setup.

klausahrenberg, 2020-07-14:
For HA please use firmware fork of @fashberg

closed 2020-07-14 by klausahrenberg