klausahrenberg/WThermostatBeca #124: Is there a changelog

IanAdd, 2020-06-29:
I downloaded a bin file a bit back version (1.08) and have just installed it on my BHT-002 using tuya-convert. 
Went like a dream, thanks so much !
Back here I see there is now version 1.11
Where can I find what the changes/benefits of doing an upgrade to this version.

fashberg, 2020-06-29:
Hi!

Until 1.10 there was one, see https://github.com/klausahrenberg/WThermostatBeca/blob/v1.10/Changes.md 

For later changes see https://github.com/klausahrenberg/WThermostatBeca/releases

Kind Regards
Folke

IanAdd, 2020-06-29:
Thanks, perfect

IanAdd, 2020-06-29:
It was a surprise that the upgrade made it forget all the settings, but all resolvable.

fashberg, 2020-06-29:
Hey @IanAdd,

starting at klaus version 1.09 network settings are kept during upgrades (1.09 is the version being upgraded).
Other settings still getting lost at some upgrades.

Kind Regards
Folke

fashberg, 2020-06-29:
@IanAdd , you're repo show some HASS stuff.
Maybe you're interested in https://github.com/fashberg/WThermostatBeca   | https://github.com/fashberg/WThermostatBeca/blob/master/Changes.md

IanAdd, 2020-06-29:
Thanks, I'm guessing you are building upon the earlier version. I will do some reading.

BTW The OTA upgrade for 1.08 straight to 1.11 lost the wifi settings too, the link you gave suggests it wouldn't.

fashberg, 2020-06-29:
> BTW The OTA upgrade for 1.08 straight to 1.11 lost the wifi settings too.

Yes, upgrading from 1.08 to 1.11 you'll loose wifi settings.
Upgrading _from_ 1.09 to greater 1.0x WiFi settings will be kept (networksettings layout changed at 1.09 and later versions are compatible).

It's different at my fork: Upgrading 1.xx-klaus to 1.15-fas Wifi Settings will be kept.
1.15-fas to 1.11(-klaus) network settings will also be kept.
1.xx-fas to 1.xx-fas: everything will be kept (when upgrading)

Kind Regards
Folke

IanAdd, 2020-06-29:
Thanks @fashberg , it looks like i need to upgrade, found gremlin.

Ok, upgraded, lots of cool new stuff, but now the clock is haywire. The link in the tasmota menu to help on such is broken.
What do I need for Sydney Australia ?

fashberg, 2020-06-29:
OK, tasmota changed it's webpage. New URL is https://tasmota.github.io/docs/Commands/#timestd

https://greenwichmeantime.com/time-zone/australia/new-south-wales/sydney/time-sydney/

StandardTime: Sydney Time Time = GMT+10
DaylightSaving: Sydney Time Time = GMT+11

Australia Daylight Saving Time begins at 2am (AEST) on the first Sunday in October and ends at 2am (AEST) (which is 3amAustralian Eastern Daylight Time) on the first Sunday in April.

DST/STD: W,M,D,h,T W = week (0 = last week of month, 1..4 = first .. fourth) M = month (1..12) D = day of week (1..7 1 = sunday 7 = saturday) h = hour (0..23) T = timezone (-810..810) (offset from UTC in MINUTES - 810min / 60min=13:30)

```
Timezone = 99    # Enable DST Calculation
STD = 1,4,1,3,600  # Starts first Sunday in April at 3 o'clock then it's UTC + 600 Minutes
DST = 1,10,1,2,660 # Starts first Sunday in October at 2 o'clock then it's UTC + 660 Minutes
```

IanAdd, 2020-06-29:
Aaah, the problem i had with 1.11 still exists.
HA sends true and false payloads out capitalised. Capitalised True is not recognised by your software. false and False both work, however.

IanAdd, 2020-06-29:
> OK, tasmota changed it's webpage. New URL is https://tasmota.github.io/docs/Commands/#timestd
>
> https://greenwichmeantime.com/time-zone/australia/new-south-wales/sydney/time-sydney/
>
> StandardTime: [...]

Great, thanks for all the effort.

closed 2020-06-30 by IanAdd