klausahrenberg/WThermostatBeca #127: Support for ME81AH.31 WIFI

Saharel001, 2020-08-07:
Hi,

I have a ME81AH.31 WIFI thermostat that registers the Smart-Life app.

Is is possible this thermostat is suppoorted? If you say possible, i try to flash my device (own risk ;) )

https://www.aliexpress.com/item/4000398480147.html?spm=a2g0o.detail.1000060.2.6a3c5b6bBkpAHG&gps-id=pcDetailBottomMoreThisSeller&scm=1007.13339.146401.0&scm_id=1007.13339.146401.0&scm-url=1007.13339.146401.0&pvid=ec184e94-db99-46da-8dd5-f7cfbb43ac64&_t=gps-id:pcDetailBottomMoreThisSeller,scm-url:1007.13339.146401.0,pvid:ec184e94-db99-46da-8dd5-f7cfbb43ac64,tpp_buckets:668%230%23131923%2362_668%23808%234093%23898_668%23888%233325%2310_668%232846%238110%23389_668%232717%237560%23222_668%231000022185%231000069217%230_668%233468%2315617%23864

lozb36, 2020-09-11:
I own a ME81H 3A WIFI Thermostat (I think its the same you have) and successfully flashed the device with the software. But at the moment I do not have any possibilities to control the thermostat via mqtt, the software also does not know the current temperature nor the targetTemperature.
I am working on this but will be thankful for any recommendations where to start, e.g. how do I get mcu commands logged?

lozb36, 2020-09-17:
I added a new Pull Request to add this new feature.
https://github.com/klausahrenberg/WThermostatBeca/pull/135

klausahrenberg, 2020-09-23:
Hi all, pull request is accepted, model supported in firmware 1.17 Thanks @lozb36

closed 2020-09-23 by klausahrenberg

jan--m, 2020-11-14:
@Saharel001 or @lozb36 have you by any chance flashed your ME81 using tuya-convert? if so, how did you get your ME81 into "ez config mode"? I've only been able to get it into "ap mode" (where it makes a SmartLife-XXXX ssid available) by holding the up+down buttons for 5 seconds while it's off, but tuya-convert doesn't support that mode =[

lozb36, 2020-11-24:
Hi, unfortanetly I was not able to use wifi for flashing. I flashed my device by soldering :-(
I tried a lot to flash the device "over the air" - but no chance.

Yocee84, 2021-05-20:
> Hi, unfortanetly I was not able to use wifi for flashing. I flashed my device by soldering :-(
> I tried a lot to flash the device "over the air" - but no chance.

Hi!

Please help me a little. I Bought 6 Me81H thermostat. I want flash with tasmota. Seriusly need unsolder the RX pin, or unsolder the whole esp? How do you flash your thermostat? Put MCU reset pin to gnd? Where is the res pin?
[image]

I flashed with unsolder half, it woket, i resolded, it worked, but when i updated tasmota it is dead. I want unsolder again, but completly broken.

rickyelopez, 2022-01-12:
Does this work with the on-board temperature sensor? Or do you have to have a remote MQTT temp sensor?

What template did you use in Tasmota for this?