klausahrenberg/WThermostatBeca #129: Can not find WNetwork.h

TwoLeftFeet, 2020-08-11:
Hi
I am just setting up an additional stat and when trying to compile in platformio I get the following error
src\WThermostat.cpp:2:22: fatal error: WNetwork.h: No such file or directory

I like to compile my own version rather than just using the .bin file.

I have looked for the file and it does not seem to be there.

I have been using the previous version ever since you released it without problem and thought as I am installing a now Stat I would try the new version.

Can you help please.....Don

fashberg, 2020-08-11:
Hi @TwoLeftFeet,

You need WAdapter library - see https://github.com/fashberg/WThermostatBeca#build-this-firmware-from-source

TwoLeftFeet, 2020-08-11:
Thank you very much...All working now.

On 11/08/2020 12:54, Folke wrote:

Hi @TwoLeftFeet<https://github.com/TwoLeftFeet>,

You need WAdapter library - see https://github.com/fashberg/WThermostatBeca#build-this-firmware-from-source

closed 2020-09-23 by klausahrenberg