klausahrenberg/WThermostatBeca #131: Firmware 1.11 Flash behavior

pewi121, 2020-08-29:
Hi Klaus,
I just updated a working BECA BHT-1000 from firmware 1.08 to 1.11. While flashing it lost all its config settings and opend the initial WIFI AP. Is that expected behavior? 
Additionally I have noticed that the idx Name is limited to 16 characters now. Is this a mistake or intent?

thank you 
Peter

fashberg, 2020-08-29:
Hi @pewi121,

every firmware has it's own config-layout and there is no conversion from old one.
Starting at version 1.09 the Networks settings-flash layout is still the same and only app-settings (clock, thermostat) changed sometimes and had to be reconfigured. My Fork has addressed this issue and can import all settings of previous versions.

So that behaviour was to be expected.

Shortening the max-length of hostname was to reduce amount of flash.
I don't no why klaus deleted the changelog.md 

https://github.com/klausahrenberg/WThermostatBeca/commit/c8539978da7eebfb5abb74fcaeb114b9d0a16f27#diff-566c8996dd07ebb891fbb6105bab6f01

Kind Regards
Folke

klausahrenberg, 2020-09-23:
> While flashing it lost all its config settings and opened the initial WIFI AP. Is that expected behavior?

Hi, the answer is: yes and no. If I implement new features with new options that have to be stored in EEPROM, then it's required that the thermostat must be reconfigured. Otherwise the thermostat will read the settings at start completely wrong, if old settings are available.
From version 1.11(~) on I implemented an improvement to minimize this not ideal reconfiguration procedure. The network settings (wifi, mqtt) will survive most of the firmware updates. Could be that other options need still a reconfiguration in future, I will mention this in the release note. However, I try to minimize this ;-)

> Additionally I have noticed that the idx Name is limited to 16 characters now. Is this a mistake or intent?

This is intended for 2 reasons: Reduce flash amount (EEPROM is limited) and the idx is used as network hostname also and should not be too long. Hope you can deal with this limitation.

closed 2020-09-23 by klausahrenberg