klausahrenberg/WThermostatBeca #132: Saswell Thermostat Valves SEA801-DF1

Sthopeless, 2020-08-30:
Hi! 

Awhile ago I purchased one of the [Saswell Gateways and TRV's](https://www.aliexpress.com/item/4000303875313.html?spm=a2g0o.cart.0.0.2d5b3c00dOzRoz&mp=1) hoping it had a ESP8266 but it's a WRG1.
I have unsoldered the [WRG1](https://fccid.io/2ANDL-WRG1/Users-Manual/User-Manual-4380962) and soldered a ESP8266 on GND, VCC, RX, TX and with Tasmota running I was able to get the [Tuya protocol messages](https://docs.tuya.com/en/iot/smart-product-solution/product-solutiongateway/gateway-mcu-access-solution/tuya-gateway-mcu-access-communication-protocol?id=K9fo8c0m7hkd0) and after some trial and error I successfully am able to send commands to the TRV's, start and stop pairing mode get the all the status..
Most the options/functions are very similar to the Thermostat controllers, please [check here](https://photos.app.goo.gl/wP14Esb4Ykw3cJPY8) to see the DpId's and the format of one of the messages I was able to capture.

I would like to have WThermostatBeca installed on the gateway controlling the TRV's but I certainly would need help or guidance.
Would this be something you're interested in? Let me know.

fashberg, 2020-08-30:
Hey @Sthopeless ,

sounds very interesting. It seems that the more popular flashing free firmware becomes the more they try to avoid them.
Read supported devices page at tasmota and you often can read "after template became available they've changed firmware and tuya-convert is not longer possible - you have to solder".
So changing the WiFi chip is also an expected answer.
Last week i've ordered a 4 pack of plugs with power metering for 26 euro and flashed tasmota using tuya-convert.

The question is: The easier flashing is the more people will use it.
Many stop at soldering. But changing the WiFi Chip is crazy and not many people will do this - my opinion.
As long as you can buy 

The serial MCU procol seems not to be ESP/WRG specific. So it's just like adding a new supported hardware.

The 801 uses Zigbee to communicate to the WiFi Unit, right?
https://www.saswell.com/etrv-smart-radiator-valve-tuya-alexa-thermostatic-radiator-valve-sea801-ty_p104.html

Here is an approach to not use the gateway but direct Zigbee2MQTT https://github.com/Koenkk/zigbee2mqtt/issues/2781

Kind Regards
Folke

rotragit, 2020-08-31:
Of course that's very interesting. In general, my perplexity (and the reason why I avoid wifi based TRV) is the power consumption that is much more than the zigbee modue (according with https://developer.tuya.com/en/docs/iot/device-development/module/zigbee-module/zigbeetyzs11module). ESP/WRG have quite similar power consumption (270mA in TX mode, 50-70mA in RX mode) while the tuya TYZS11 module has 8mA in RX mode and a range from 11.8mA to 120mA depending on the distance from the gateway.
How long is expected to work with batteries, even with the original WRG1 module?
This problem doesn't affect plugs or switches connected to the main power, of course (apart radio pollution in the house ^^ )

Sthopeless, 2020-08-31:
Hi and thank you for your replies!!

@fashberg generally you can do some snooping with a ftdi before unsoldering the module, I've had great success transplanting the modules I have done it aswell to 2 Vacuum cleaners which are now running Tasmota.

Saswell has 2 versions for this model one is indeed Zigbee and the other one (mine) are RF 878mhz https://photos.app.goo.gl/z5UUAUo9qfR5qpW77
I have other TRV's that are zigbee and integration with Zigbee2MQTT isn't the best atm.

I bought this Saswell gear before this Zigbee boom, the TRV units are RF only and don't contain any wifi module therefor the battery life is the same or better than zigbee versions. The Gateway had a WRG1 and it's powered with a adapter, I've sniffed and made sense of the protocol and I have it running on Tasmota the TRV's are added to HA aswell with a very long yaml file for each.

klausahrenberg, 2020-09-23:
> I would like to have WThermostatBeca installed on the gateway controlling the TRV's

Sounds interesting, but for time reasons I will not be able to implement new devices, sorry.
Furthermore, I think, this wouldn't be an easy task. This seems to be a controller for one or MORE control units at each heater. If this device can read, e.g. the temperature of each heater separatly - than I would have to implement this reading/control of different units. I would recommend to implement a separate firmware for this device to use all features.

closed 2020-09-23 by klausahrenberg