klausahrenberg/WThermostatBeca #133: flash from tasmota

Yoh-Zic, 2020-09-14:
good day! is there a way to flash from tasmota?

fashberg, 2020-09-14:
Hey @Yoh-Zic !

check out my forks doc: https://github.com/fashberg/WThermostatBeca/blob/master/Flashing.md#option-3-flashing-on-top-of-tasmota-or-other-free-firmware

Yoh-Zic, 2020-09-14:
your advice helped, thank you! but there is a new problem(((
what should I do if my device disappears from the network completely after entering the wi-fi settings?

fashberg, 2020-09-14:
Hi @Yoh-Zic 

have you read and tried this: https://github.com/fashberg/WThermostatBeca/blob/master/Configuration.md#network-recovery

Yoh-Zic, 2020-09-15:
this method stopped working as soon as I installed third party firmware

klausahrenberg, 2020-09-23:
> this method stopped working as soon as I installed third party firmware
I can only explain for my firmware (but it should be similar for fashberg fork): If you press the "down" button until the display blinks, the thermostat will open an own WiFi-AccessPoint for configuration, if no connection is available. So you will be able to recover the thermostat, even if you put in the wrong WiFi settings at first time.

closed 2020-09-23 by klausahrenberg