klausahrenberg/WThermostatBeca #136: work app

JD99, 2020-09-23:
Hello. Please answer the questions. 
1. Does the application (tuya smart) work after flashing? 
2. Does the device control (buttons) work after the firmware? 
3. Is it possible to return the original firmware? 
Sorry for bad English.

fashberg, 2020-09-24:
Hey @JD99 ,

> Does the application (tuya smart) work after flashing?

Of course, not! That's why there is an open source alternative

> Does the device control (buttons) work after the firmware?

sure. No differences here. 

> Is it possible to return the original firmware?

Only if you have made a backup and only with direct serial connection (soldering or esp pogo adapter).

Folke

klausahrenberg, 2020-09-24:
@fashberg answered it all

closed 2020-09-24 by klausahrenberg