klausahrenberg/WThermostatBeca #137: Schedules / Clock | Pages not working 1.16 + 1.17

mf-1, 2020-09-29:
Firmware 1.16 + 1.17 ... Schedules tab not working.

[image]

Firmware 1.17 ... Clock tab empty ...

[image]

GoNzCiD, 2020-09-29:
I have the same issue

klausahrenberg, 2020-09-30:
Strange... Could be, that loading of old settings is the cause. Please try to unzip and flash the attachment. This firmware will reset the settings of thermostat (network settings not). Please try, if this fixes the faulty clock page.

[WThermostat_1.17_reset_settings.zip](https://github.com/klausahrenberg/WThermostatBeca/files/5303665/WThermostat_1.17_reset_settings.zip)

mf-1, 2020-09-30:
@klausahrenberg Have used another thermostat (1.11) to update ... Same issue. If I reload the page the output is different. Sometime more "data", sometimes not... Clock sometime is working, sometime blank.

I've just seen that the html tag is used twice. Also seen (partial) at my screen above.
Please find attached source code from schedules page...

[view-source_IP_schedules_.zip](https://github.com/klausahrenberg/WThermostatBeca/files/5304682/view-source_IP_schedules_.zip)

klausahrenberg, 2020-10-03:
Ok, I was able to reproduce the failure, if I change the thermostat model to BAC-002-ALC for example. Cause not known yet - give me time to check the issue

mf-1, 2020-10-03:
Sounds great. Maybe this helps. I'm using BHT-002GALW

klausahrenberg, 2020-10-04:
It's all about memory...
AsyncWebServer is faster and can handle multiple requests - but memory usage increases also. This caused corrupted html sites in some cases. It's fixed now in v1.18

closed 2020-10-04 by klausahrenberg

mf-1, 2020-10-04:
acknowledged .... It's working with 1.18. Tested on another v1.11 I had.... thx!