klausahrenberg/WThermostatBeca #138: Clock problem

tibo888, 2020-10-04:
Hello,
The clock does not show the correct time. on the other hand mqtt is correct

Configuration clock
- Use fixed offset to UTC time -> "epochTimeFormatted":"2020-10-04 07:31:47" CORRECT
- Get time zone via internet -> "epochTimeFormatted":"2020-10-04 07:41:38" CORRECT

But clock time from the thermostat -> 23H02 NOT CORRECT

Ps: Guys, you doing a great job
[image]

kibermanis, 2020-10-04:
@tibo888  please check you setup. With firmware v1.18 works fine. 
[image]

klausahrenberg, 2020-10-04:
Hi, I checked it at my model (BHT-002-GBLW) and its working fine with 'timezone via internet' and/or 'fixed offset'. I need more information. Which firmware version (v1.18)? Which model? Did you received messages like 'unknown MCU command' in MQTT?

UlrichMai, 2020-10-05:
@tibo888, your picture is showing a red center touch button. Is that really red or did the camera just get the colors wrong?

tibo888, 2020-10-05:
Hello everyone,
@kibermanis i check the setup. i install the firmware v1.18 yesterday (previsouly 1.11) and the same setup. the problem is still there.
@klausahrenberg, this is the information
Model BHT-002-GBLW
No MCU error
@UlrichMai  the center touch button is orange

thanks for help

tibo888, 2020-10-06:
Hello
I found something interesting
I have 2 thermostats that are working properly and others that are not

WORK
[image]

DONT
[image]

1300371, 2020-11-09:
Hello, I'm using the zigbee version BHT-002-GCLZ with zigbee2mqtt. 
I'm facing the same issue. Did you find any workaround ?
I also ask my question here: https://github.com/Koenkk/zigbee2mqtt/issues/4185#issuecomment-724245233

Gulpman, 2021-02-03:
I do have the same issue: The clock of my thermostat ME81(A)H / firmware 1.20l is not set correctly via ntp.
Reads/writes via MQTT are working. 
My NTP settings are exactly the same as in the screenshot from @kibermanis 
No MCU Error messages in the MQTT output.

1300371, 2021-02-10:
The issue seems resolved on z2m side with the tuya reimplementation. 
See https://github.com/Koenkk/zigbee2mqtt/issues/4185#issuecomment-771608497