klausahrenberg/WThermostatBeca #139: option 'lib_deps' in section 'env:esp01_1m' already exists'

mf-1, 2020-10-05:
@klausahrenberg Using PlatformIO v 5.0.1 I will get this message about lib_deps. Ist there another way to implement `WAdapter`?

```
.../WThermostat$ platformio --version
PlatformIO, version 5.0.1

.../WThermostat$ platformio run      
Error: Invalid '.../WThermostatBeca/WThermostat/platformio.ini' (project configuration file):
    'While reading from '.../WThermostatBeca/WThermostat/platformio.ini' [line 21]:
    option 'lib_deps' in section 'env:esp01_1m' already exists'

```

fashberg, 2020-10-13:
hey @mf-1 

Klaus is coding in Atom and his repo has .gcc-flags.json containing:
```"gccIncludePaths": "/mnt/Daten/Users/klaus/Projects/Atom/WThermostatBeca/WThermostat/include,/mnt/Daten/Users/klaus/Projects/Atom/WThermostatBeca/WThermostat/src,/home/klausahrenberg/.platformio/packages/framework-.....```
 
Are you using VSCode?

Should be ok to just  checkout WAdapter directly into WThermostat\lib\WAdapter
Then change includes to:
#include <WAdapter/WDevice.h>
#include <WAdapter/WNetwork.h>

You can also try my fork, which is HASS (HomeAssistant) and VisualStudioCode optimized - the WAdapter is included as git submodule, see https://github.com/fashberg/WThermostatBeca

Kind Regards
Folke

mf-1, 2020-10-14:
@fashberg . I'm using PlatformIO, too ... Have changed something (don't remember me, sorry) .. now compiling is working.

mf-1, 2020-10-14:
Just for information. "my" current `platformio.ini`

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp01_1m]
platform = espressif8266
board = esp01_1m
framework = arduino
board_build.flash_mode = dout
board_build.ldscript = eagle.flash.1m.ld
upload_resetmethod = nodemcu
upload_speed = 921600
lib_deps =
	ESP8266WiFi
	https://github.com/me-no-dev/ESPAsyncWebServer
	AsyncTCP
	ESP8266mDNS
	PubSubClient
	DNSServer
	EEPROM
	NTPClient
	Time
	Hash

```

closed 2020-10-14 by mf-1