klausahrenberg/WThermostatBeca #143: Single Values Problem with quotation marks

esalii, 2020-10-17:
Hi,

Single values payloads have quotation marks for example shedules mode = "auto" or "off" and state = "heating" or "off"

and it is not possible to work with these values in openHab. Can you please fix this?

Thanks a lot

klausahrenberg, 2021-01-24:
Hi, I don't understand it exactly. OpenHab is not working, when these enumerations are in quotation marks? But, if I delete the quotation marks and that are not numbers or boolean values - then I have a mal-formatted json-payload, isn't it? I'm working with node-red and there it shows me errors for this kind of json payloads.

Could you explain your requirement more in detail, please?

lozb36, 2021-01-25:
@esalii , it should work with openhab.
My things file (if you use files) looks like this:

`Thing topic thermostat_whz "Thermostat Wohnzimmer" {`
`    Channels:`
`        .....`
`        Type switch : T_WHZ_HEATING                "T WHZ"                     [ stateTopic="thermostat_whz/thermostat/properties/systemMode",`
`                                                                                 commandTopic="thermostat_whz/thermostat/set/systemMode", on="\"heat\"", off="\"cool\"" ]`
`        Type switch : T_WHZ_SCHEDULE               "T WHZ Schedule"            [ stateTopic="thermostat_whz/thermostat/properties/schedulesMode",`
`                                                                                 commandTopic="thermostat_whz/thermostat/set/schedulesMode", on="\"auto\"", off="\"off\"" ]`
`        Type switch : T_WHZ_DEVICEON               "T WHZ DeviceOn"            [ stateTopic="thermostat_whz/thermostat/properties/deviceOn", commandTopic="thermostat_whz/thermostat/set/deviceOn", on="true", off="false"]`
`    }`

As you could see, I added escaped quotation marks to the strings with quotation marks. Hope this is helpful.

klausahrenberg, 2021-01-26:
Thanks @lozb36 for explain the issue

closed 2021-01-26 by klausahrenberg