klausahrenberg/WThermostatBeca #144: BHT-002-GCLW board layout

tbekavac, 2020-10-24:
Hi guys, I bought BHT-002-GCLW but the board is not like the one from the configuration documentation. 
How can I do the HW modification on it (is it needed now)? 
And what about flashing? You'll notice that now there are holes marked with TDIO, TCK, GND & 3.3V.
[image]

kocsishome, 2020-10-30:
Hello!

Is the temperature data OK on the web interface?

tbekavac, 2020-11-13:
Hi! To check this I would need to flash the firmware? 
I wanted to check what is possible before I do anything with the device...

klausahrenberg, 2021-01-23:
> It has a little different dps config:
> 
> 1: True / False - on/off state
> 2: 26 - setpoint
> 3: 25 - sensor temperature
> 4': 'manual' / 'week' / 'eco' - mode
> 7: True / False - lock status
> 11: '0' - ?
> 12: '0' - ?
> 13: '0' - ?
> 14: 'warming' / 'heating' - relay state
> 19: 'c' - ?

If someone is interested to test. I implemented the support for DLX_LH01 in my firmware according to the list above. in my version 1.20i. Needs further testing and 'warming'/'heating' not impl. yet.

closed 2021-02-20 by klausahrenberg