klausahrenberg/WThermostatBeca #145: DST (daylight saving time) not switching back?

rrednaxela, 2020-10-26:
After falling back from DST to normal time on Sunday morning the time on my BHT-002-GALW didn't adjust. Today around 20:00 I clicked save on the 'Configure clock' page, and after the reboot the time was ok.
I have configured the defaults, pool.ntp.org and time zone server http://worldtimeapi.org/api/ip
How often does the software update the time?
I have 2 thermostats on 1.18, same behavior on both.

fashberg, 2020-10-26:
Klaus pulled that part of my fork and there was a bug in my Time Zone Implementation (mixed Northern and Southern Hemisphere).

Check https://github.com/fashberg/WThermostatBeca/issues/62 and 
https://github.com/fashberg/WThermostatBeca/commit/274a38e005c03d417e9893de388c6ef10d01b92f

fashberg, 2020-10-26:
Nope. Completely different here and no Northern/Southern Mismatch https://github.com/klausahrenberg/WThermostatBeca/blob/master/WThermostat/src/WClock.h#L515
Then something different.

@rrednaxela You shoud share a screenshot of your TimeZone Settings

rrednaxela, 2020-10-27:
> @rrednaxela You shoud share a screenshot of your TimeZone Settings
just the defaults
[image]

Provided that the time was correct after reboot, it seems the condition in line 129 is not met? Or it never gets there?
I looked a bit at 
https://github.com/klausahrenberg/WThermostatBeca/blob/4e76dfcdd6c806b34d6e25b9c61f4a6413d78874/WThermostat/src/WClock.h#L105
+ lines 112, 129:
How much is 60000? 1 minute (according to comment in L104)?

What is then the point of " (now - lastTimeZoneSync > 60000)" in L129? Only update timezone if last timezonesync was more than 1 min ago?  If the whole loop anyway runs only every 3 hours?

martin072, 2022-04-02:
Bit of an old topic, but found out the same, but the other way around. DST settings were not applied after the 27th of March. I changed the NTP server to nl.pool.ntp.org, and after a reboot the time was correct again.

helmar74, 2023-10-29:
Same here today with version 1.36.
Clock didn't change with this settings:
[image]