klausahrenberg/WThermostatBeca #148: targetTemperature set by the software is not saved after power 220V = "OFF"

YuriT1, 2020-10-28:
Hello!
In  firmware (1.18): after disconnecting the device from the power supply 220V (OFF) , and ==> turning on the power 220V (ON), targetTemperature = the last target temperature set on the device, but not programmatically set (for example, on the device we set the temperature 20 degrees with our hands. Then we send it to the mqtt topic == >> targetTemperature = 25. The topic publishes targetTemperature = 25. Disconnect the thermostat from 220V power == >> Next, connect the thermostat to 220V power. The topic publishes targetTemperature = 20 !!!) 
Will this issue be resolved in the next firmware?

bmw-biker, 2021-01-21:
Yes, would benice to have.

For me it seems to be an extra feature just for the BHT-002 series. My BHT-002GALW also looses its settings in the thermostate itself. Even a brand new one. From the original manual I would read that the settings are stored - in the cloud. They say that settings get restored - if you connect the device to the cloud.

This is what I do currently. If the settings don't match whith my FHEM I do an update via mqtt.
I haven't tried publishing as retained data yet - maybe this could also be a solution ?

My Tuya X7H thermostate stores its settings also on a power cycle.

Best regards

klausahrenberg, 2021-01-23:
Checked with my devices. It seems a special disadvantage of BHT-002 device. My other device (ME102H) stores the changed values via MQTT:

Description of bug: All changes at parameters that were modified over Wifi and not on the device itself are lost after power interruption, e.g. deviceOn, targetTemperature or schedules. If you change it on the device, all changes are stored at power loss. 

This is really annoying because the Wifi has not enough memory to store all parameters twice. At the moment there is no good solution for this problem.

pashok2398, 2021-01-23:
@klausahrenberg is it possible to disable the schedule to free up the memory for the state?

klausahrenberg, 2021-01-24:
Ok, here we go. I still have an inner resistance to solve a device bug... I also own a BHT-002, I have not so much power interruptions, but anyhow - this behavior of device is not acceptable for me also.
In version 1.20j I did a ugly implementation to solve this issue. I ask everyone to test, if this works.
At the moment only property 'deviceOn' is modified for testing. Please test the solution:
* switch the device on or off via MQTT
* Interrupt the power supply
* Switch power supply on: The thermostat should go in the On/Off State that you selected before via MQTT

Any confirmation that this solution in v1.20j works is welcome. Then I will proceed for other properties (targetTemperature, schedules, ...) and will do this in a more clean coding style.

wongcz, 2021-02-01:
Hi, it seems that deviceOn state is not saved. I have turned off BHT-002 via MQTT, cut off power and when the power was restored the thermostat switched on.

hartmut58, 2021-04-05:
Hi, I'm testing BHT-002 with v1.20 and its works fine.

Vadim1623, 2023-09-05:
Hello!
Good job!
But targetTemperature still not saved after shutdown.