klausahrenberg/WThermostatBeca #149: ThermostatBeca New version DLX Control solution

alessandrosalafia, 2020-10-30:
Hi Folke. I recently ordered new home thermostats. They arrived and I found they have different hardware. I'm still 8266 and I was able to load your firmware, but there are problems in displaying and exchanging data.
[image]

kocsishome, 2020-10-31:
[image]

tbekavac, 2020-11-13:
Hi! Same thing here :)
Duplicate of #144

klausahrenberg, 2021-01-23:
I implemented support for this new device DLX-LH01 in version 1.20i. Please test and report the unknown MCU commands. I will close this issue because it's a duplicate of #144 .

closed 2021-01-23 by klausahrenberg