klausahrenberg/WThermostatBeca #160: MQTT command case sensitive

EugenePanyaev, 2020-11-28:
Hello. I am trying to change state of AVATTO ME81H and there is a problem:

Publish message - **False** to thermostat1/thermostat/set/deviceOn - working
Publish message - **false** to thermostat1/thermostat/set/deviceOn - working
Publish message - **true** to thermostat1/thermostat/set/deviceOn - working
Publish message - **True** to thermostat1/thermostat/set/deviceOn - **NOT** working

This is part of HA config:

  - platform: mqtt
    name: thermostat1_state
    icon: mdi:lock
    state_topic: "thermostat1/thermostat/properties/deviceOn"
    value_template: '{{ value_json }}'
    state_on: true
    state_off: false
    command_topic: "thermostat1/thermostat/set/deviceOn"
    payload_on: true
    payload_off: false

**true** is used and **True** is sent by HA and as a result, the device cannot be turned on

Same issue with locked property. True - not working, true - working

Firmware 1.8

Sorry for English :-)

Polomans, 2020-12-14:
How did you manage to flash this device? tuya-convert not work, soldering method doesn't work either. 
I saw your photo on Ali, I solder it the same way, but it does not connect ...

EugenePanyaev, 2020-12-14:
I did it according to the instructions, at first it didn't work, I checked
the contacts and everything worked out. I have 3 more devices, I'll try
them later

lozb36, 2020-12-29:
How do you activate mqtt messages with unknown MCU commands?

EugenePanyaev, 2020-12-29:
> How do you activate mqtt messages with unknown MCU commands?

I didn’t do anything except according to the instructions. The temperature on the thermostat changes, the temperature from the sensor is transferred to the HA.

klausahrenberg, 2021-01-10:
This change will com TruE at next version. So far only 'true' values are accepted. Wait to next version 1.20 - then it will be fixed.

klausahrenberg, 2021-01-18:
Fixed since 1.20. Notice, the version needs some further testing - still alpha status.

closed 2021-01-18 by klausahrenberg