klausahrenberg/WThermostatBeca #162: Support for HY02B05-WIFI

yvesdm3000, 2020-12-02:
Hello,

Could you give me some pointers to support the Tuya HY02B05-WIFI Thermostat?
It has an esp based TYWE3S but a PIC16F1947 chip as MCU. Since only VCC,GND, GPI15, RX and TX are connected, most functions will certainly be perfomed by the MCU.
As I understand, Tuya has some standard way of interfacing with the serial, so I assume this will also be the case? 

Any pointers in how you solved the support for your existing devices so I can have a jumpstart at supporting this thermostat ?

Thanks,

Yves

yvesdm3000, 2020-12-02:
[image]
[image]

yvesdm3000, 2021-01-04:
OK I now flashed tasmota, saves much time to find out the protocol.

This is what I now gathered:

[image]

I guess I'll have to press some buttons and see what changes now, document it and maybe add it to WThermostat ?

yvesdm3000, 2021-01-05:
After some testing:

DpId 1: Boolean,  Switches the device on or off, display clears, I hear the relay clicking when it's heating
DpId 2: Integer,  target temperature x 10 so 240 is 24.0 degrees C
DpId 3: Current measured temperature x 10, so 215 is 21.5 degrees C
DpId 4: Enum, 
    0 = Manual, 
    1 = Auto, 
    2 = Bag icon, 
    3 = Manual override I think, when updating DpId2, this switches to 3
    4 = Disabled I think (no state icon displayed)
    5 = Disabled I think (no state icon displayed)
DpId 6: Boolean, Display lock
DpId 101: Boolean, function unknown
DpId 102: Boolean, function unknown
DpId 103: Integer, function unknown
DpId 104: Integer, function unknown
DpId 105: Integer, function unknown
DpId 106: Boolean, function unknown
DpId 107: Boolean, function unknown
DpId 108: Boolean, function unknown
DpId 109: Integer, function unknown
DpId 110: Integer, function unknown
DpId 111: Integer, function unknown
DpId 112: Integer, function unknown
DpId 113: Integer, function unknown
DpId 114: Integer, function unknown
DpId 115: Integer, function unknown
DpId 116: Enum, function unknown
DpId 117: Enum, function unknown
DpId 118: Enum, function unknown

klausahrenberg, 2021-01-13:
Hi, please select HY08WE as model for your thermostat. Following commands fits already to your model Target/Measured temperature and display lock. There should be some unknown MCU messages reported to your mqtt broker. Please report this.

klausahrenberg, 2021-01-14:
Check binary WThermostat_1.20b_alpha, I added support for schedules mode (DpId 4). Should be Manual, Auto, Holiday, Hold (acc. to manual: hold temp for 1 h, and then switch back to auto?). Please check

yvesdm3000, 2021-02-03:
> 
> 
> Hi, please select HY08WE as model for your thermostat. Following commands fits already to your model Target/Measured temperature and display lock. There should be some unknown MCU messages reported to your mqtt broker. Please report this.

I now updated the firmware to 1.20l_beta. I still need to add it to MQTT to see if it's functional on that side, but so far it looks good.

Yves

yvesdm3000, 2021-02-03:
I have these popping up continuously:

{"error":"Unknown MCU command: 55 aa 01 07 00 05 0c 05 00 01 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 65 01 00 01 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 66 01 00 01 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 67 02 00 04 00 00 00 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 68 02 00 04 00 00 00 01"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 69 02 00 04 00 00 00 05"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 6a 01 00 01 01"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 6b 01 00 01 01"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 6c 01 00 01 01"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 6d 02 00 04 ff ff ff f6"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 6e 02 00 04 00 00 00 0a"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 6f 02 00 04 00 00 00 02"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 70 02 00 04 00 00 00 2d"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 71 02 00 04 00 00 00 05"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 72 02 00 04 00 00 00 23"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 73 02 00 04 00 00 00 05"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 74 04 00 01 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 75 04 00 01 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 76 04 00 01 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 0c 05 00 01 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 65 01 00 01 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 66 01 00 01 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 67 02 00 04 00 00 00 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 68 02 00 04 00 00 00 01"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 69 02 00 04 00 00 00 05"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 6a 01 00 01 01"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 6b 01 00 01 01"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 6c 01 00 01 01"}

yvesdm3000, 2021-02-03:
{"notice":"MCU: 55 aa 01 07 00 05 01 01 00 01 01"}
{"notice":"MCU: 55 aa 01 07 00 05 04 04 00 01 01"}
{"notice":"MCU: 55 aa 01 07 00 08 02 02 00 04 00 00 00 64"}
{"notice":"MCU: 55 aa 01 07 00 08 03 02 00 04 00 00 00 b4"}
{"notice":"MCU: 55 aa 01 07 00 05 06 01 00 01 00"}
{"notice":"MCU: 55 aa 01 07 00 05 0c 05 00 01 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 0c 05 00 01 00"}
{"notice":"MCU: 55 aa 01 07 00 05 65 01 00 01 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 65 01 00 01 00"}
{"notice":"MCU: 55 aa 01 07 00 05 66 01 00 01 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 66 01 00 01 00"}
{"notice":"MCU: 55 aa 01 07 00 08 67 02 00 04 00 00 00 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 67 02 00 04 00 00 00 00"}
{"notice":"MCU: 55 aa 01 07 00 08 68 02 00 04 00 00 00 01"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 68 02 00 04 00 00 00 01"}
{"notice":"MCU: 55 aa 01 07 00 08 69 02 00 04 00 00 00 05"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 69 02 00 04 00 00 00 05"}
{"notice":"MCU: 55 aa 01 07 00 05 6a 01 00 01 01"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 6a 01 00 01 01"}
{"notice":"MCU: 55 aa 01 07 00 05 6b 01 00 01 01"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 6b 01 00 01 01"}
{"notice":"MCU: 55 aa 01 07 00 05 6c 01 00 01 01"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 6c 01 00 01 01"}
{"notice":"MCU: 55 aa 01 07 00 08 6d 02 00 04 ff ff ff f6"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 6d 02 00 04 ff ff ff f6"}
{"notice":"MCU: 55 aa 01 07 00 08 6e 02 00 04 00 00 00 0a"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 6e 02 00 04 00 00 00 0a"}
{"notice":"MCU: 55 aa 01 07 00 08 6f 02 00 04 00 00 00 02"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 6f 02 00 04 00 00 00 02"}
{"notice":"MCU: 55 aa 01 07 00 08 70 02 00 04 00 00 00 2d"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 70 02 00 04 00 00 00 2d"}
{"notice":"MCU: 55 aa 01 07 00 08 71 02 00 04 00 00 00 05"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 71 02 00 04 00 00 00 05"}
{"notice":"MCU: 55 aa 01 07 00 08 72 02 00 04 00 00 00 23"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 72 02 00 04 00 00 00 23"}
{"notice":"MCU: 55 aa 01 07 00 08 73 02 00 04 00 00 00 05"}
{"error":"Unknown MCU command: 55 aa 01 07 00 08 73 02 00 04 00 00 00 05"}
{"notice":"MCU: 55 aa 01 07 00 05 74 04 00 01 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 74 04 00 01 00"}
{"notice":"MCU: 55 aa 01 07 00 05 75 04 00 01 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 75 04 00 01 00"}
{"notice":"MCU: 55 aa 01 07 00 05 76 04 00 01 00"}
{"error":"Unknown MCU command: 55 aa 01 07 00 05 76 04 00 01 00"}
{"notice":"MCU: 55 aa 01 00 00 01 01"}

yvesdm3000, 2021-02-03:
I don't have any mqtt topic in '#' that shows me the current temp, nor the set temp.

yvesdm3000, 2021-02-04:
Regarding the unknown DpID values, another user (SergyiSviderskii) on tasmota gave me this information:

DpId 102: Boolean-relay output (load)
DpId 103: Integer- external sensor temperature
DpId 104: Integer-number of vacation days
DpId 105: Integer-vacation temperature
DpId 106: Boolean-high temperature protection was cancelled ( A6 - [--] - customization HY02B05)
DpId 107: Boolean-low temperature protection was cancelled ( A7 - [--] - customization HY02B05)
DpId 109: Integer- measure temperature calibrtion -9/+9 ( A1 - customization HY02B05)
DpId 110: Integer- temperature control return difference setting 0.5/2.5 ( A2 - customization HY02B05)
DpId 111: Integer -external sensors limit temperature control return difference 1/9 ( A2 - customization HY02B05)
DpId 112: Integer-limit value of high temperature for external sensor ( A6 - 35°C/70°C - customization HY02B05)
DpId 113: Integer-limit value of low temperature for external sensor ( A7 - 1°C/10°C - customization HY02B05)
DpId 114: Integer-setting temperature highest limit 20°C/70°C ( A9 - customization HY02B05 - max temp. )
DpId 115: Integer-setting temperature lowest limit 1°C/10°C ( A8 - customization HY02B05 - min temp. )
DpId 116: Enum-options of sensor control N1,N2,N3 ( A4 - customization HY02B05 )
DpId 117: Enum-power with memory function 0-memory,1-off,2-on ( AB- customization HY02B05 )
DpId 118: Enum-weekly programming selection 0-5+2, 1-6+1, 2-7 ( AC- customization HY02B05 )
and
DpId 119,DpId 120,:string- 6 -boundary points working days or week 7 days
DpId 121,DpId 122:string- 6 -boundary points weekend