klausahrenberg/WThermostatBeca #163: Schedules empty, cannot add/edit ...

mf-1, 2020-12-09:
Strange issue. Have seen that one of my WT's has lost scheduled temperature setting.

[image]

Have entered all data again and saved. Device has rebooted but schedules settings remains empty.

mf-1, 2020-12-09:
Have "updated" to v1.16 - same issue... have set 23 to weekday period 1 and save. After reboot the field remains "0.0".

[image]

psxde, 2020-12-11:
Do you have set '23' or '23.0' ? I had problems on initial setup, too but it worked over mqtt. Now I can change values also in webgui but I have used decimal numbers.

mf-1, 2020-12-11:
Tried both ... 23.0 and 23 .. same result. Will try to set via MQTT later this day.

mf-1, 2020-12-27:
Wanted to try today... It seems that all Beca devices will not connect to MQTT since a long time ... because of retain I can see that some were last connected at 2020-12-09 // 2020-12-04 // 2020-09-29 (yes,Sept)

Checking the thermostat it-self schedules seem there ... but the UI isn't showing this ...

mf-1, 2020-12-29:
I've found the issue ...

My MQTT broker is using Ethernet and WiFi with different IPs, of course.
DNS resolves both IPs ... no problem...

I have used FQHN within WT.
But WiFi connection was cut since a long time, I've never noted this before, and DNS has resolved both IP's furthermore.

In this case WT was using the WiFi IP for MQTT connection. But there was no MQTT broker.

After changing devices to wired IP it was working and all missing schedules were available again ...
To approve this behavior I've changed some temperatures ...
* devices rebooted, changes values were available.
* changed MQTT host back to FQHN, reboot, schedules empty (temp:"0" / zero).
* changed back to IP it was working again. Changed values were still present.

At this point I've checked Mosquitto. At this point I've seen that WiFi connection wasn't available.
Bad thing. DNS resolves both IP furthermore. :-( / FritzBox 7490, even the connection was offline.

The reason for that `wlan0` was offline, was the firmware from Raspberry OS. After updating the firmware was handle `wlan0` again and WiFi was also available.

Changed some devices back to FQHN and it's working furthermore. Alse schedules are available ... as expected.

@klausahrenberg
The behavior is a little bit strange for me. If MQTT broker isn't available schedules tab is empty. Please, can you have a look at this, if possible.

klausahrenberg, 2021-01-18:
which thermostat model?

mf-1, 2021-01-18:
> which thermostat model?

I'm using BHT-002-GALW

klausahrenberg, 2021-01-18:
Ok, now I got it. I already have an idea. The thermostat state is requested after a mqtt connection is established. No mqtt connection, no request and no schedules. Seems like a bug, I will check.

klausahrenberg, 2021-01-20:
Thanks for your hint with the mqtt broker. This issue is fixed now at v1.20f. Please try

closed 2021-01-20 by klausahrenberg

mf-1, 2021-01-20:
It's working ... have set an IP where no broker is and schedules were fully visible as expected. :)

Btw... there is a new message .. 

> Subscribe to topic 'devices/#' at your broker to get device information.

But there are no messages :)
Maybe because of I've set an individual mqtt topic.

klausahrenberg, 2021-01-20:
At my mqtt broker 2 messages arrive at topic `devices/clock` and 'devices thermostat`:

```json
{"url":"http://192.168.1.xx/things/clock","ip":"192.168.1.xx","stateTopic":"thermostatbeca/clock/properties","setTopic":"thermostatbeca/clock/set"}

```json
{"url":"http://192.168.1.xx/things/thermostat","ip":"192.168.1.xx","stateTopic":"thermostatbeca/thermostat/properties","setTopic":"thermostatbeca/thermostat/set"}

But thermostat should be connected to a mqtt broker for this - which is not, if you reproduced the schedules issue above :-)