klausahrenberg/WThermostatBeca #165: MQTT Status

psxde, 2020-12-11:
Hi,

a Online/Offline LWT value in MQTT would be nice to detect whether the device is online/connected or not.

Maybe this can be added in the next version.

Thanks,
Patrick

klausahrenberg, 2021-01-24:
Sorry, I don't get the idea. What exactly is the request? An online and offline status via MQTT? How should the device communicate the 'offline' state via MQTT - if it's not connected? I think, I have a wrong idea of your request here :-) Please clarify

psxde, 2021-01-25:
Hi Klaus, 
exactly - an online and offline state. If you search for MQTT LWT you should find details about LWT (last will and testament). In short: You can store a last will message in the mqtt broker which will then be sent after the client is disconnected ungracefully.

klausahrenberg, 2021-01-25:
Ok, cool idea - I din't knew this. It's already implemented and works great. Will come in version 1.20k, but before publishing I will fix some other issues. Just wait a little. I solved this and putted an 'alive' boolean value in the device state report. The LWT message contains the idx, ip and "alive":false in payload.

klausahrenberg, 2021-01-30:
implemented in 1.20k

closed 2021-01-30 by klausahrenberg