klausahrenberg/WThermostatBeca #171: ME81AH - not able to set any settings

J-rethinkmx, 2020-12-24:
I’m not able to set any values (like targetTemperature or schedulesMode) via http, MQTT, or direct code changes. I see the serial commands being sent, however the next query response has them set to the original values. Manual mode is clearly set on the display. Any changes on the display are reflected in received serial commands.

I’m sure this worked before, however may have done something via playing with the code or even on the display itself. Have switched back to the baseline WThermostatBeca.bin and still the same result.

any suggestions?

klausahrenberg, 2020-12-30:
I will have a look at this in new year. Maybe I have to reorganize the firmware to reflect all the different models with different commands. ME81AH was implemented by @lozb36 , maybe he has any suggestions for this behavior? However, this model has different schedules with only 8 time slots instead of 18. I also have a similar model now - but not time at the moment. Will check next year

lozb36, 2020-12-30:
I guess I know the error.
The software sends commands beginning with:
55AA**0006**0005
But the thermostat needs (ME81AH):
55AA**0307**0005
I will try to check it during the next days, but at the moment I am quite busy.

But: I am able to send updates for the target Temperature. This is the only thing I currently use.

J-rethinkmx, 2021-01-01:
Hey guys thanks for chiming in and taking a look.

I have addressed the issues. Basically there were serial communications collisions. It seems the MEH81 needs time to process a command and can’t handle back to back commands without waiting for the response. The set target temp method sends the temp command and a set schedule mode to manual command and that usually always fails. In addition a handshake command wake up cycle can coincide with a command being sent (as it’s coming asynchronously via a http or MQTT session). The device loop logic is such that a reset of the last handshake cycle (via setting to millis() after sending a command) doesn’t work as loop takes a “now” value which is much earlier than the new millis value and results in a negative/large interval period which yields the handshake command being sent right away.

I have added states to handle all these sends, wait for response, and sent multiple commands cases and more and it works without incident.

Also the MEH81 handle/support the wifi status/reset through GPIOs instead of messages, which I added support for. As well I have identified and handled all but 1 or 2 of the unknown MCU commands coming back via the query command.

The setSystemMide method also doesn’t use the Id array fir its command, although that was just discovered when playing/debugging.

I’ll get that code cleaned up and centralized in the next few weeks and get it posted.

bmw-biker, 2021-01-07:
Hi guys, 

I'm very new to Klaus' thermostat firmware. Just for test I ordered two different thermostats via aliexpress: 
BHT-002-GALW [https://de.aliexpress.com/item/32905181466.html?spm=a2g0s.9042311.0.0.3da24c4doylG05](url)
X7H Water Wifi [https://de.aliexpress.com/item/1005001597908456.html?spm=a2g0s.9042311.0.0.3da24c4doylG05](url)

I can confirm BHT-002-GALW and Tuya X7H working.
Thanks for the great work !  Must have been a lot...

BHT-002-GALW is working perfect via MQTT and Web-UI with model BHT_002_GBLW.

With X7H I have similar issue as J-rethinkmx with ME81AH.
I tried all models. The one working best (nearly perfect) is MK70GB-H.
From device all properties are transmitted correctly via MQTT.
I also tried to set the properties via MQTT - multiple in one Json struct, to ease the test. I changed one and checked the result. I thought that setting properties doesn't work at all.
By accident I tried to send one single property as JSON. This way I noticed that every property can be set correctly - most times.
Sometimes I had to repeat the command.

I assume the handshake issue described by J-rethinkmx. I would really be interested in your solution.

Happy new year and stay healthy

J-rethinkmx, 2021-01-18:
Hey @bmw-biker i should be pushing something up mid week. It works nicely and i need to cleanup some stuff.

Also @klausahrenberg, what is the best strategy to add a new settings value to the networks settings? Do I have to create a new version/level and provide conversion or does the new value just get added to the end? ... thanks

klausahrenberg, 2021-01-18:
Hi all,
I just bought a new model (ME102H) and I think the implementation of the support will address/fix some issues with ME81AH also:

> targetTemperature

There is a bug in v1.19. If you set the target temperature, it will not generate a direct MQTT response. this will be fixed soon. 

> Also the MEH81 handle/support the wifi status/reset through GPIOs

This will be supported, same at the model ME102H.

However, the handshake issue I cannot follow. Maybe I understand it wrong. But I can't imagine, that there is a timing/handshake problem in serial communication. The firmware is not waiting for a response within a specified time frame or something similar. The firmware wait for responses from MCU, whenever the MCU is answering. But I had one strange issue with ME102H device - and maybe this problem has your device too: I did not asked the MCU for the WIFI status and GPIO configuration, there are special MCU commands for this. And as long as I didn't asked the device, the ME102H device had kind of lags during tuching the screen and response of device. So I had to implement the original order of MCU commands - then this kind of delay was gone.

Long story short: Please wait for the next version (1.20d), maybe I can finish it today.

@J-rethinkmx 

> what is the best strategy to add a new settings value to the networks settings

which kind of settings do you want to implement? With version 1.20.. I changed the programming objects to make support for new devices easier. Device ME81AH has now it's own class. Hope this helps to implement support of device specific features in a more simple manner. Tell me, which kind of new setting value do you want to implement - then I give you instructions.

klausahrenberg, 2021-01-18:
By the way, I also recognized during adding support for ME102H, that the implementation for schedules for devices with only 8 time slots (instead of originally 18) is faulty. The firmware crashes, if you try to set the schedules via web interface. It will be fixed also from 1.20d on.

J-rethinkmx, 2021-01-18:
@klausahrenberg I saw the crash problem with schedules some time ago and as it wasn't an issue with what i immediately needed so just ignored it.

I really like the subclass reorg you have done with 1.20. I was going to do that once i got my code all cleaned up. I grabbed the latest version for code compares as part of that cleanup and was rejoiced. 

The timing issue was primarily due to the loop logic and to be honest probable sloppy MCU handling of the ME81. A setTarget would be initiated via a HTTP Put and then (in most cases) a handshake command via loop (as the HTTP handling logic was slow and drove the "need a handshake now logic") so they were streamed back to back to the MCU where it wouldn't respond or didnt see the setTartget command string..... a bit of execute/wait/done state logic and all was good. Its been operational for a week or so along with GPIO and some other changes and it works great. Ill wait for you next update to test and if still some issues will update accordingly.

As part of the current settings reset between the MCU/Wifi a restart in AP mode and reconfig was not happening. Needed to start into this AP/restart mode so added a bool setting in network to force loadsettings to be false. This way it happened immediately. the existing logic was sufficient.

Thanks for all your hard work on this.... much appreciate by many !

klausahrenberg, 2021-01-20:
Schedules issue is fixed since 1.20e - now ESP doesnt crash when setting on devices with only 8 instead of 18 time slots. Please try.

> As part of the current settings reset between the MCU/Wifi a restart in AP mode and reconfig was not happening. Needed to start into this AP/restart mode so added a bool setting in network to force loadsettings to be false. This way it happened immediately. the existing logic was sufficient.

Should I add an option "Reset device"? This option would erase all settings (Wifi and device settings) - so it would be immediatly available as an AP?

J-rethinkmx, 2021-01-20:
Ill try the E version, although not until the weekend at the earliest and report back on the schedules and timing issues

On the reset issue, my implementation kept the settings intact as many of them were still valid and it was more about the wifi settings that were wrong. This was also further complicated as GPIOs were used instead of messages. 

A "reset" bool setting based on the hard resetting of the device would be ideal. This would restart the device in AP mode... Further a button to erase all values and then have the code set this "reset" setting and restart once again.

bmw-biker, 2021-01-21:
Many thanks for all your tries. I've been very busy lately. No time for test so much.

I also recognized the crash probably due to the schedules on my X7H thermostate (V1.20c). The crash caused me to have it serially flashed. For a quick solution I went back to 1.18 for the X7H.
I already mounted it and don't want to unmount, open and solder again. So I would like to wait for my next order for a test. They say it comes till midth of Feb.
For now the X7H with V1.18 works pretty well. I only change one property at a time... ;-)
Only, with its pretty short update intervall it seems that the thermostate itself has a small bug: mqtt sometimes shows a wrong temperature - but I also noticed this behaviour on the display from the corner of my eye.

The BHT-002-GALW works like a charme with V1.20c. Also here - I order some new devices and will do tests with those in mdth of Feb.

lozb36, 2021-01-21:
I tested 1.20d so far with mqtt only.

I was able to change temp, switch thermostat on/off. I also received temperature update messages when I change the target temperature at the thermostat. 
With mqtt I receive changed schedules modes with quotation marks like "off".
It was not possible for me to change the schedules mode auto/off via mqtt. But this maybe corresponds to the error above.
Changing schedules via web worked well.

I also had some trouble with building from source because of different methods in WAdapter project.

I really like the new architecture with separate classes for each type. Great work. Thanks a lot @klausahrenberg

klausahrenberg, 2021-01-23:
> A "reset" bool setting based on the hard resetting of the device would be ideal. This would restart the device in AP mode... Further a button to erase all values and then have the code set this "reset" setting and restart once again.

I added 2 options in latest beta:
* 'Restart in AP mode' > reboots the ESP and starts AccessPoint. There you can overwrite the existing settings.
* Reset all settings > Delete network and device settings. Complete reconfiguration is needed

closed 2021-01-30 by klausahrenberg