klausahrenberg/WThermostatBeca #172: HY09-GCWRW support?

Stoatwblr, 2020-12-27:
https://www.aliexpress.com/item/4000689067944.html

This apears to be a more or less standard Tuya device driving a water heating boiler. I've actually flashed the -fas fork but I don't think this is fas specific

I cannot get it to join my wifi network (SSID is 10-chars mixed case alphanumeric, WPA2/3 pass 25 chars mixed case alphanumerica plus punctuation (?!"$%^&*(){}][#~@, etc)

After providing the SSID and password it reboots, loops a coupe of times and then falls back to AP mode

This isn't a general tasmota problem as my Tasmota devices are working fine

The wifi unit is a TYWE3S in the relay head unit

These can't be flashed OTA with tuya-convert but they can be flashed with usb-ttl serial and grounding gpio0 + disabling the 433MHz transceiver

Stoatwblr, 2020-12-27:
Hopefully this helps

file:///tmp/20201227_004125.jpg

Stoatwblr, 2020-12-27:
I've got an annotated image of the control board, but can't upload it.

Stoatwblr, 2020-12-27:
Same faiiure using WThermostat_1.18.bin (without fallback to AP mode that fas offers)
I can see it periodically trying to attach to the Wifi network, but the Fritzbox isn't logging failures so I don't think it's even touching the right SSID

Stoatwblr, 2020-12-27:
Solved the networking problem:

 (for some reason a space was being appended to the SSID on my android phone browser and I missed this. I'm leaving this here for anone else who might get caught out. It took 3 tries to clear this field on the browser)

Stoatwblr, 2020-12-27:
NTP sync is working and transmitted to the controller over the 433MHz link

Set as HY08 device

Temperatures reporting OK (no floor temp)
no relay/load status
wifi "factory settings" button on front panel doesn't work

BOILER-SWITCH = {"error":"Unknown MCU command: 55 aa 00 07 00 05 76 04 00 01 00"}

the only connection between the TYWE3S and the rest of the unit is via TX/RX pins

Stoatwblr, 2020-12-27:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

klausahrenberg, 2020-12-30:
Ok, thanks for report for this device. I will have a look at this in new year. Maybe I have to reorganize the firmware to reflect all the different models with different commands.

tarasifua, 2021-06-19:
Hello! Any progress with that model?

Stoatwblr, 2023-08-02:
None yet

I have contacted Hysen and asked for an API for the MCU, in order to get canonical data on DPids. No response yet

Stoatwblr, 2024-06-03:
DPID table (I'll edit as I find more)
``
all vals seem to be 4 byte and hex (not bcd)

0x01 dpid 1                             - display enabled
0x02 dpid 2 val(4)                 - target temp x100 (0.1C step)
0x03 dpid 3 val(4)                 - actual temp x100 (0.1C step)
0x04 dpid 4 enum 0,1,3     - mode 0=manual, 1=schedule, 3=schedule+override
0c06 dpid 6 bool - always 0? 
0x0c dpid 12 bitmap - always 0?
0x65 dpid 101 bool off -  always ?
0x66 dpid 102 bool              -  heat relay (switch)
0x67 dpid 103 val(4) - always 0?
0x68 dpid 104 val(4)  - always 1?
0x69 dpid 105 val(4) - always 0x10/16
0x6a dpid 106 bool off - always ?  on now
0x6b dpid 107 bool on - always ?
0x6c dpid 108 bool on - always ?
0x6d dpid 109 val(4)            - "A1" int sensor offset temp (C x 10) 
0x6e dpid 110 val(4)            - "A2" int sensor temp tolerance (C x 10)
0x6f dpid 111 val(4)            - "A3" ext sensor temp tolerance (C x 1)
0x70 dpid 112 val(4)            - "A5" max temp of external sensor (C x 1) - under 35 = disabled
0x71 dpid 113 val(4)            - "A6" Min temp of internal sensor (frost protect) (C x 1) - over 10 = disabled val 5 - always
0x72 dpid 114 val(4)            - "A8" Maximum settable tmperature
0x73 dpid 115 val(4)            - "A7" Minimum settable temperature             
0x74 dpid 116 enum 00 - always?
0x75 dpid 117 enum 00           - "AA" Power off recovery 0=resume 1=off 2=on
0x76 dpid 118 enum 00           - "AB" 00=P1(5+2), 01=P2(6+1), 02=P3(7 days)
0x79 dpid 121 string - 00:00:09:06:00:14:08:00:0F:0B:1E:0F
0x7A dpid 122 string - 00:00:09:0D:1E:0F:91:00:0F:16:00:0F
```

Stoatwblr, 2024-06-05:
Judging how the DPids match the HY08, I'm willing to bet that the unknowns are similar

On the other hand it's interesting how the divisors are different as are the scheule dpids

Stoatwblr, 2024-06-05:
Another update (I'm using esphome for this at the moment)

23:12:09][C][tuya:041]: Tuya:
[23:12:09][C][tuya:056]:   Datapoint 1: switch (value: ON)
[23:12:09][C][tuya:058]:   Datapoint 2: int value (value: 150)
[23:12:09][C][tuya:058]:   Datapoint 3: int value (value: 215)
[23:12:09][C][tuya:062]:   Datapoint 4: enum (value: 1)
[23:12:09][C][tuya:056]:   Datapoint 6: switch (value: OFF)
[23:12:09][C][tuya:064]:   Datapoint 12: bitmask (value: 0)
[23:12:09][C][tuya:056]:   Datapoint 101: switch (value: OFF)
[23:12:09][C][tuya:056]:   Datapoint 102: switch (value: OFF)
[23:12:09][C][tuya:058]:   Datapoint 103: int value (value: 0)
[23:12:09][C][tuya:058]:   Datapoint 104: int value (value: 3)
[23:12:09][C][tuya:058]:   Datapoint 105: int value (value: 8)
[23:12:09][C][tuya:056]:   Datapoint 106: switch (value: OFF)
[23:12:09][C][tuya:056]:   Datapoint 107: switch (value: ON)
[23:12:09][C][tuya:056]:   Datapoint 108: switch (value: ON)
[23:12:09][C][tuya:058]:   Datapoint 109: int value (value: 0)
[23:12:09][C][tuya:058]:   Datapoint 110: int value (value: 15)
[23:12:09][C][tuya:058]:   Datapoint 111: int value (value: 5)
[23:12:09][C][tuya:058]:   Datapoint 112: int value (value: 40)
[23:12:09][C][tuya:058]:   Datapoint 113: int value (value: 10)
[23:12:09][C][tuya:058]:   Datapoint 114: int value (value: 30)
[23:12:09][C][tuya:058]:   Datapoint 115: int value (value: 5)
[23:12:09][C][tuya:062]:   Datapoint 116: enum (value: 0)
[23:12:09][C][tuya:062]:   Datapoint 117: enum (value: 0)
[23:12:09][C][tuya:062]:   Datapoint 118: enum (value: 0)
[23:12:09][C][tuya:054]:   Datapoint 119: raw (value: 04.1E.14.08.00.0F.0B.1E.0F (9))
[23:12:09][C][tuya:054]:   Datapoint 120: raw (value: 0D.1E.0F.11.00.0F.16.00.0F (9))
[23:12:09][C][tuya:054]:   Datapoint 121: raw (value: 06.00.14.08.00.0F.0B.1E.0F (9))
[23:12:09][C][tuya:054]:   Datapoint 122: raw (value: 0D.1E.0F.11.00.0F.16.00.0F (9))
[23:12:09][C][tuya:074]:   Product: 'qpbczz4wwv6q0mxu1.0.0'

It's interesting to see DPIds 119/120 appear. This beings it to match the Hysen HY02 and it seems they're only emitted _very_ infrequently or when schedules are changed