klausahrenberg/WThermostatBeca #174: Case-insensitive MQTT payload check

dimerr, 2021-01-04:
Hello,

by default HomeAssistant sends payload for power_command_topic with capital letter and it isn't possible to customize that.
Turns out that:

`mosquitto_pub -t warmfloor/thermostat/set/deviceOn -m "True"`
is not the same as:
`mosquitto_pub -t warmfloor/thermostat/set/deviceOn -m "true"`

```
warmfloor/thermostat/set/deviceOn True
warmfloor/thermostat/set/deviceOn True
warmfloor/thermostat/set/deviceOn true
warmfloor/thermostat/properties {"idx":"warmfloor","ip":"192.168.1.47","firmware":"1.18","temperature":20.50,"targetTemperature":26.00,"deviceOn":true,"schedulesMode":"off","locked":false,"floorTemperature":0.00,"systemMode":""}

```

While 
```
warmfloor/thermostat/set/deviceOn False
warmfloor/thermostat/properties {"idx":"warmfloor","ip":"192.168.1.47","firmware":"1.18","temperature":20.50,"targetTemperature":26.00,"deviceOn":false,"schedulesMode":"off","locked":false,"floorTemperature":0.00,"systemMode":"heat"}

```
works just fine...

pashok2398, 2021-01-06:
Duplicate #160

klausahrenberg, 2021-01-10:
This change will com TruE at next version. So far only 'true' values are accepted. Wait to next version 1.20 - then it will be fixed.

klausahrenberg, 2021-01-18:
Fixed since 1.20. Notice, the version needs some further testing - still alpha status.

closed 2021-01-18 by klausahrenberg