klausahrenberg/WThermostatBeca #175: Lost settings - targetTemperature

fabitom, 2021-01-04:
I'm on BHT-002 (floor heating) with 1.18
I've configured thermostat with MQTT and all works nice except targetTemperature. When I set as example 23 then after couple minutes/hours thermostat reset this value to 20.
I also see some problems when I change targetTemperature by LCD touch screen, sometimes it reset to 15. There must be some issue with synchronization.

pashok2398, 2021-01-06:
@fabitom have the same problem, i notice that its happens because it goes to schedulesMode: "auto", when i move it to manual by click or schedulesMode: "off" by MQTT its back to normal.
Not sure why but suddenly it switch itself to "auto":

```
{"idx":"thermostat_01","ip":"192.168.1.113","firmware":"1.18","temperature":25.00,"targetTemperature":20.50,"deviceOn":true,"schedulesMode":"off","ecoMode":false,"locked":false,"floorTemperature":73.50,"state":"heating"}

{"idx":"thermostat_01","ip":"192.168.1.113","firmware":"1.18","temperature":25.00,"targetTemperature":15.00,"deviceOn":true,"schedulesMode":"auto","ecoMode":false,"locked":false,"floorTemperature":73.50,"state":"heating"}
```
This is from my MQTT explorer...

pashok2398, 2021-01-06:
Found it....
```
Auto mode from manual mode at next schedule period change
(not at model ET-81W and ME81AH)
```
Remove the checkbox in http://<YOUR_IP>/thermostat?

fabitom, 2021-01-06:
@pashok2398 thanks, unchecked auto mode. Testing... It can be that option. I will write result after a couple hours ;)

fabitom, 2021-01-07:
Yes, all is good now. Sorry that was auto mode option.

closed 2021-01-07 by fabitom