klausahrenberg/WThermostatBeca #176: Version Error 1.19 vs 1.16

helmar74, 2021-01-09:
Is there a version error in the repo?
[image]

klausahrenberg, 2021-01-09:
No, it's a typo from me in my commit message.

This is version 1.19 definitely

helmar74, 2021-01-09:
So I can update my Thermostats?
Are there any breaking changes?

klausahrenberg, 2021-01-09:
No breaking changes between 1.18 and 1.19. You can update, no functional changes, only improved reliability.

helmar74, 2021-01-09:
Thanks for the info.
Updated my 8 GBLW's to Version 1.19:
[image]

Thank you for your good work!!!!!

closed 2021-01-09 by helmar74