klausahrenberg/WThermostatBeca #178: WIFI Key

alex-fra, 2021-01-18:
tested: BHT-6000GBLW-W with version 1.19 and 1.20a
Wifi key is shortened to 30 characters. My key with over 60 characters is therefore not working.

klausahrenberg, 2021-01-18:
Hi, Wifi passphrase of more than 30 chars was never supported. Sorry to say, but because the passphrase is stored in EEPROM and the space there is limited to only 512 byte - I can not add 30 more bytes for the password. The space is almost completely used by current settings already.

How about to create a guest network with a shorter password and limited access to your network structure?

closed 2021-01-18 by klausahrenberg

alex-fra, 2021-01-18:
A shame, that would be so perfect.
Thank you for your work.
I have to check whether it is possible to get access to my home network with the guests wifi of the Fritz box